platformio/platformio-core #441: raspberrypi espressif board problem

nge88, 2016-01-07:
Hi,

I've successfully installed platformio on raspberry pi (running on raspbian, I've tried both with Wheezy and Jessie) but I cannot install  espressif boards:

>**platformio platforms install espressif**
Installing toolchain-xtensa package:
Downloading  [####################################]  100%              2%  00:03:44
Unpacking  [------------------------------------]    0%
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 326, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 113, in install
    if self.unpack(dlpath, pkg_dir):
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 52, in unpack
    return fu.start()
  File "/usr/local/lib/python2.7/dist-packages/platformio/unpacker.py", line 103, in start
    self._unpacker.extract_item(item, self._dest_dir)
  File "/usr/local/lib/python2.7/dist-packages/platformio/unpacker.py", line 36, in extract_item
    self._afo.extract(item, dest_dir)
  File "/usr/lib/python2.7/tarfile.py", line 2107, in extract
    self._extract_member(tarinfo, os.path.join(path, tarinfo.name))
  File "/usr/lib/python2.7/tarfile.py", line 2183, in _extract_member
    self.makefile(tarinfo, targetpath)
  File "/usr/lib/python2.7/tarfile.py", line 2224, in makefile
    copyfileobj(source, target)
  File "/usr/lib/python2.7/tarfile.py", line 265, in copyfileobj
    shutil.copyfileobj(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 49, in copyfileobj
    buf = fsrc.read(length)
  File "/usr/lib/python2.7/tarfile.py", line 824, in read
    buf += self.fileobj.read(size - len(buf))
  File "/usr/lib/python2.7/tarfile.py", line 742, in read
    return self.readnormal(size)
  File "/usr/lib/python2.7/tarfile.py", line 751, in readnormal
    return self.fileobj.read(size)
  File "/usr/lib/python2.7/gzip.py", line 268, in read
    self._read(readsize)
  File "/usr/lib/python2.7/gzip.py", line 315, in _read
    self._read_eof()
  File "/usr/lib/python2.7/gzip.py", line 350, in _read_eof
    crc32 = read32(self.fileobj)
  File "/usr/lib/python2.7/gzip.py", line 25, in read32
    return struct.unpack("<I", input.read(4))[0]
error: unpack requires a string argument of length 4

============================================================

Any thoughs?
Thank you 

nge88

ivankravets, 2016-01-07:
@valeros could you repack Espressif toolchain for ARM?

valeros, 2016-01-07:
Hi @nge88!
Please update package via `platformio update`
Thanks!

nge88, 2016-01-08:
Hi @valeros 
Thank you, now platform espressif is installed; it even compile the project (via **platformio run**)
But if I try any upload to board i get this error:

>sudo platformio run --target upload
[Fri Jan  8 08:45:46 2016] Processing esp12e (platform: espressif, board: esp12e, framework: arduino)
--------------------------------------------------------------------------------
Installing tool-scons package:
Already installed
Installing toolchain-xtensa package:
Already installed
Installing tool-esptool package:
Already installed
Installing tool-mkspiffs package:
**Error: The package 'tool-mkspiffs' is not available for your system 'linux_armv6l'**

Thank you so much
nge88

valeros, 2016-01-08:
Sorry @nge88, `tool-mkspiffs` for ARM was missed.
Please try to upload firmware again! Thanks!

nge88, 2016-01-08:
Ok it's working!!
Thank you for your fast support

closed 2016-01-08 by nge88