platformio/platformio-core #444: Not clear how to overwrite $BOARD.options.resetmethod ..

seclorum, 2016-01-08:
I would like to force my board.options.resetmethod="ck", but see no way to do that outside of editing the board.json file .. is there a way to override this for my board in a project-local config?  (i.e. through platformio_extra_script.py - tried to override it like "esp12.board.options.resetmethod=ck" but no go ..

valeros, 2016-01-08:
Hi @seclorum!
How about create [custom board](http://docs.platformio.org/en/latest/platforms/creating_board.html) based on `esp12` with needed reset method?

seclorum, 2016-01-12:
I'd really rather not do that, since it means maintaining another set of file(z) outside the project repository - rather would like to know how to over-ride the defaults for this key:

````
    "esp12e": {
        "build": {
            "core": "esp8266",
            "extra_flags": "-DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12",
            "f_cpu": "80000000L",
            "ldscript": "esp8266.flash.4m.ld",
            "mcu": "esp8266",
            "variant": "nodemcu"
        },
        "frameworks": ["arduino"],
        "name": "Espressif ESP8266 ESP-12E",
        "platform": "espressif",
        "upload": {
            "maximum_ram_size": 81920,
            "maximum_size": 4194304,
            "resetmethod": "ck",
            "require_upload_port" : true,
            "speed": 115200
        },
        "url": "http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family",
        "vendor": "Espressif"
    },
````

ivankravets, 2016-01-12:
@seclorum how about http://docs.platformio.org/en/latest/projectconf.html#extra-script ?

ivankravets, 2016-01-13:
Please use http://docs.platformio.org/en/latest/projectconf.html#extra-script

`platformio.ini`
```ini
[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
extra_script = extra_script.py
```

`extra_script.py`
```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()
env['BOARD_OPTIONS']['upload']['resetmethod'] = 'ck'
```

ivankravets, 2016-01-13:
See http://docs.platformio.org/en/latest/platforms/espressif.html#custom-reset-method

ivankravets, 2016-01-13:
Implement `upload_resetmethod` option for http://docs.platformio.org/en/latest/projectconf.html#uploading-options

Ping @crushedice2000

ivankravets, 2016-01-13:
Please re-test with the latest http://docs.platformio.org/en/latest/installation.html#d-development-version

ivankravets, 2016-01-13:
Docs: http://docs.platformio.org/en/latest/platforms/espressif.html#custom-reset-method

closed 2016-01-13 by ivankravets

crushedice2000, 2016-01-14:
@ivankravets: Thank you very much!

commit b60eb059 (2016-01-13): Explain in documentation how to overwrite upload reset method for Espressif development platform // Resolve #444
commit 52ceb647 (2016-01-13): Added "upload_resetmethod" option to Project Configuration File platformio.ini and allowed to change default upload reset method for Espressif development platform // Resolve #444