platformio/platformio-core #446: Templates engine & Exmples

kireevco, 2016-01-09:
I would love to see templates in the projects, so when you init a project you could init it with an example, that will have certain information filled out, for example:

`platformio init -b megaatmega2560 --ide clion --template BasicSerial` will init Arduino basic serial project, so you don't have to copy-paste things.

Also, what can be done, is implementing jinja2 templating enging to replace certain values in the actual code (for example serial baud rate:)

```
void setup() {
    Serial.begin({{serial_baudrate}});    
}
```

Where `serial_baudrate` could be a parameter in the PIO global file or local (by the precedence).

Please let me know what do you guys think.

ivankravets, 2016-01-13:
Good idea, thanks! Need to understand that these examples depend on framework type too.

mogorman, 2016-01-14:
i like idea about serial_baud_rate, would be better better though ifi it was just passed via #defines though.  although how those are set could be done in the platformio.ini file for sure

skorokithakis, 2016-04-19:
I would also love this. Ability to pull from a git repo directly would be great, but it's not too much hassle for me to just pull the repo somewhere and use it as a template locally.

DrSensor, 2018-02-11:
<details>
<summary>this is just a quick comment, I will add the detail later</summary>

Integrating with cookiecutter maybe will be a good idea so anyone can make their own custom project structure (example: arduino project with reactjs).

I notice that this case has some similiarity with `vue invoke` in vue-cli@3.0. For feature injecting example into main code stated in first comment, it need to parse AST of C language (if for a moment this feature limited only to framework that use C++ like Arduino). Python lib that I know support parsing AST of cpp source is pycparse (although its not really well documented, there is https://github.com/eliben/pycparser/tree/master/examples for starter). Because many arduino lib provide example source code, the feature code can be extracted from there then inject it to the main code. 
</details>

DrSensor, 2018-02-28:
I think for template feature, it can be split into 2 kinds of template.

## 1. Scaffold Template

This template is a custom template for a very specific case. For example, template project for building Web Server in ESP8266 that will consist of Arduino code and SPA Framework specific code (Angular/Vue/React). However, Scaffold Template is a feature just to facilitate if anyone wants to make their own template project (like in the corporate when they need to integrate their Prototype with their own Service).

### Usage

```cmd
pio init gh:username/platformio-mytemplate
```

This will fetch template from that GitHub repo then generate the files and folder from that.

### Technical Suggestion

Luckily in python, there is a command-line utility that creates projects from project template called [cookiecutters](https://cookiecutter.readthedocs.io/en/latest/case_studies.html). I think it's best if it builds on top of that to reduce the complexity and make the documentation more simple by linking it to cookie-cutter documentation. Also, it can be [called from python](https://cookiecutter.readthedocs.io/en/latest/advanced/calling_from_python.html)

---
## 2. Template from Library Example

This feature is more like *injector* rather than _template_. Some Library (especially many Arduino Library) has Code Example like in [here](https://platformio.org/lib/show/2071/AWS_IOT) that can be treated as a Template then it can be analyzed then injected into the current project file.

### Usage

```bash
pio example <lib_name> --list
```

#### Basic

```bash
pio lib example <lib_name> --print <example_name>                       # print code example to console
pio lib example <lib_name> --print <example_name> --output <filename>   # output code example to file
```

#### Inject

```bash
pio lib example <lib_name> --add
pio lib example <lib_name> --add <example_name>
```

For example, if we want inject/add the implementation of DHT11\_Logger with Library AWS\_IOT into the current project:

```bash
pio lib example AWS_IOT --add DHT11_Logger
```

<details>
<summary>command that's not necessery to implement</summary>

This command is not necessary to implement because better to delete it manually rather than automatically (as long as it's injecting into one file).

```cmd
pio example <lib_name> --remove
pio example <lib_name> --remove <example_name>
```

</details>

### Technical Suggestion

This feature is a little bit complex to implement, especially in inject usage. To implement that, it needs to parse [AST](https://en.wikipedia.org/wiki/Abstract_syntax_tree) of the example code then inject it into the main code (some of the javascript project generators usually use this technique). Python lib that I know support parsing AST of C++ source is [pycparse](https://github.com/eliben/pycparser) (although it's not really well documented, there is [code example](https://github.com/eliben/pycparser/tree/master/examples) for the starter). Also, note that the Example code of the Library is Framework and Platform specific. I think the simplest Framework to experiment this feature is Arduino.

AlesSt, 2022-11-29:
I would like to add my vote to this request/plea also as I would REALLY love to see this feature added.

I have about 3 or 4 different base "templates" that I usually start all my projects with (and usually always forget something hence my hairline thinning of all the head scratching where did I go wrong :) ).

So my suggestion/need would be:

Templates: 
  - consists usually of custom platformio.ini file (mainly my own "half-libs") 
  - also custom partition.csv file 
  - and custom main.cpp file

As @DrSensor already mentioned years ago about Scaffold  Template (project init with different template setups from git) would would be a spot on feature.

Otherwise ABSOLUTELY AMAZING work by PIO crew. So I would like to THANK YOU ALL that are working on this project that we all greatly benefit from.