platformio/platformio-core #448: Platform IO Bundled Distribution

kireevco, 2016-01-09:
As some Windows user might not have any console tools installed I would like to see platformio distributed as some self-contained archive/exe file for Windows users. It should contain python distribution and some nice colorful command line + GUI (which is in the works). This should be done side-by-side to existing pip installation or even on top of it.

Please let me know what you think

alonewolfx2, 2016-01-09:
+1 thank you for suggestions. I hope someone can make it.

ZachMassia, 2016-01-10:
I think [Nuitka](http://nuitka.net/) might be worth looking into for this. 

From what I've read, it can't do cross-platform compilation, so someone with a Windows PC would have to give it a shot.

The following command from the PlatformIO root dir should produce a completely portable `.exe` which would work on a fresh Windows install, no external dependancies. 

`nuitka --recurse-all --standalone ./platformio`

I've tried it on my Mac and can confirm that it does compile without errors. Don't mind the `.exe`, Nuitka appends it to the generated executable on all operating systems.

```
git clone https://github.com/platformio/platformio
cd platformio

nuitka --recurse-all --standalone ./platformio
[compile output not shown]

cd platformio.dist
./platformio.exe --version
PlatformIO, version 2.7.2.dev0
```

ivankravets, 2016-01-11:
Thanks @ZachMassia. I didn't know about Nuitka. 

1. I see that users will have problems that need to add path of binary to ENV.PATH
2. The `upgrade` process will be so difficult. The `platformio upgrade` looks more interesting.

Windows users don't like to play with console utilities. If user can download and install pre-compile `platformio.exe`, then it will not have problems to `pip install platformio`.

My propositions:

1. If Windows user wants to use PlatformIO as CLI, then will install it in generic way (via `pip` or installer script http://docs.platformio.org/en/latest/installation.html#b-installer-script).
2. We are working on the own solution (IDE, or plugins for IDE) where user will "DOWNLOAD & USE" and PlatformIO will be included in IDE/Plugin.

ivankravets, 2016-01-24:
I've released beta version [PlatformIO IDE for Atom](http://docs.platformio.org/en/latest/ide/atom.html) today. It contains built-in PlatformIO CLI tool and Terminal. Also, users have ability to install this built-in PlatformIO CLI tool globally using `Menu: PlatformIO > Install Shell Commands`, then open system Terminal and `platformio` and `pio` should be available.

gepd, 2016-02-06:
@ZachMassia I've tryied what you described but I didn't get the same results.

First try:
1. I clone the platformio repository in the desktop
2. nuitka --recurse-all --standalone ./platformio

Start to compile and shows me a lot of warnings like `cannot find 'package name' as relative or absolute import`

when the compilation finished, I run `platformio.exe --version` and the output is `ImportError: No module named _socket`

Second try:
1. I compile from `C:\Python27\Lib\site-packages\platformio`

Same warning outputs

But when I run `platformio.exe --version`
It shows me the version I've installed

when I move the `platformio.dist` folder to other location I get the error `ImportError: No module named _socket` again

You know what can I be doing wrong?

gepd, 2016-02-06:
[cx_Freeze](https://cx-freeze.readthedocs.org/en/latest/overview.html) ([Download](http://cx-freeze.sourceforge.net/)) Seems to be a better alternative because it doesn't "hide" the source files, so the upgrade option should work. (In theory) and it's available to create a msi installer for windows and dmg for OSX.

I've compile the code but I still have to know how to including all the platformio files (code). 
With the basic [setup.py](https://cx-freeze.readthedocs.org/en/latest/distutils.html#distutils) configuration, are available the commands `lib`, `platforms`, `upgrade`

When I run `__main__ --version` the output is: `PlatformIO, version 2.8.3` (the name `__main__` is easy to fix, only have to edit two files, but until now I'm trying to get work the whole code)

* `lib` seems to work fine

* `platforms` the command seems to work but isn't possible to search or install any platform

* `upgrade` the output is `Error: Can not retrieve the latest PlatformIO version`

I think it's because the missing files, Is there someone with any experience using cx_Freeze?

gerritv, 2016-03-03:
All Windows version in recent history have cmd.exe available very easily. Just Shift/Right click on a directory in Windows Explorer and click on Open Command Window Here.