platformio/platformio-core #451: Specficy upload port for Mac AND windows ?

donnib, 2016-01-11:
Hi,
Is there a way to specify an upload port both for mac and windows ? I work both on windows and on mac on the same project. 

donnib

ivankravets, 2016-01-11:
Please take a look at http://docs.platformio.org/en/latest/projectconf.html#extra-script

```python
from platform import system
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

if system().lower() == "windows":
    env.Replace(UPLOAD_PORT="COM13")
else:
    env.Replace(UPLOAD_PORT="/dev/tty.serialport")
```

ivankravets, 2016-01-11:
Sorry, why do you need to specify directly `upload_port`? PlatformIO can automatically find upload port. Try `platformio run -t upload` without specifying `upload_port` or `--upload-port`.

ivankravets, 2016-01-19:
Please reopen issue if you need help.

P.S: We would be thankful for a star :star:  https://github.com/platformio/platformio/stargazers

closed 2016-01-19 by ivankravets