platformio/platformio-core #452: Programming Arduino Yun over wifi?

grzegorzshub, 2016-01-12:
Is it possible to program Yun over wifi please?

ivankravets, 2016-01-13:
I hope yes. Please explain me how does it look from CLI.

grzegorzshub, 2016-01-14:
I don’t know I'm afraid. I use Atom with Platformio and program Yun over USB only.
I have found: https://www.twilio.com/blog/2015/02/arduino-wifi-getting-started-arduino-yun.html
In the second half of the page there is description how to set Arduino IDE with Port over IP and program it over the air.
Just wondering it there is a way to do it with platformio.
Thank you

ivankravets, 2016-01-14:
Please enable verbose mode for building and uploading in Arduino IDE settings. Then run uploading via OTA and share here link with output http://pastebin.com

grzegorzshub, 2016-01-19:
done
http://pastebin.com/1kGkPVK9

ivankravets, 2016-01-19:
TODO: Make uploading via SSH https://github.com/arduino/openwrt-packages-yun/blob/master/arduino/yun-scripts/files/usr/bin/run-avrdude

AKBON, 2016-04-12:
Hi,

what is the status of this feature? Am using deviot which offers me to manually add an IP under "Serial Port(s)", but if I trigger the upload, it only says "None serial port selected".

My System is: Windows 7, Sublime + Deviot, Arduino Yun connected via LAN

ivankravets, 2016-04-12:
Hi,

Sorry for delay. We will add full support for Arduino Yun as soon as possible.

> Hi,
>
> what is the status of this feature? Am using deviot which offers me to manually add an IP under "Serial Port(s)", but if I trigger the upload, it only says "None serial port selected".
>
> My [...]

Regards, Ivan Kravets
   - Founder and CTO at PlatformIO.Org
   - Ph.D, Researcher and Software Architect
   - Creativity comes from talent and never from knowledge (c)
________________________
http://www.ikravets.com

yveshanoulle, 2016-10-09:
last reaction to have it soon was april . We are now October. Is there any progress?

ivankravets, 2016-12-04:
Moved to https://github.com/platformio/platform-atmelavr/issues/23

closed 2016-12-04 by ivankravets