platformio/platformio-core #453: SPL-Framework support for Nucleo F401RE board

g7uvw, 2016-01-12:
A request to add SPL framework support for the Nucleo F401RE board. CMSIS works well, but there are some higher level features in SPL that would be very nice to have access to.

closed 2016-01-13 by valeros

ivankravets, 2016-01-13:
Please install development version http://docs.platformio.org/en/latest/installation.html#d-development-version

g7uvw, 2016-01-14:
I'll try it out first thing tomorrow.

thanks.

g7uvw, 2016-01-19:
Sorry - only just had time to get to test this.
It looks like the framework-spl is missing some files.
It's definitely missing stm32f4xx_hal.h and stm32f4xx_hal_rcc.h
My build gives up at that point.

valeros, 2016-01-19:
Hi @g7uvw these files `stm32f4xx_hal.h` and `stm32f4xx_hal_rcc.h` belong to ST HAL library.

g7uvw, 2016-01-24:
Ok, I think I'm misunderstanding something.
I thought the SPL was built upon the HAL. I'll have to go away and read some more.

commit e2d9d0a5 (2016-01-13): Add SPL support for nucleo_f401re // Resolve #453