platformio/platformio-core #455: error: previous declaration of 'void __run_user_rf_pre_init()' with 'C++' linkage

crushedice2000, 2016-01-13:
I'm having a strange problem with PlatformIO. When I try to build the following code I get the above error:

     #include <ESP8266WiFi.h>
     #include <ESP8266mDNS.h>
     #include <WiFiUdp.h>
     #include <ArduinoOTA.h>
     #include <ESP8266WebServer.h>
     #include <ArduinoJson.h>
     #include <DNSServer.h>
     #include <FS.h>

     #define TXMODE 2   // {'PHY_MODE_11B': 1, 'PHY_MODE_11G': 2, 'PHY_MODE_11N': 3}
     #define TXPOWER 0  // TX power in dBm
     #define TXCHANNEL 7

     #define MACADDR {0x00, 0x00, 0x01, 0xED, 0xA1, 0xBA}

     // StaticJsonBuffer<200> jsonBuffer; // DynamicJsonBuffer jsonBuffer;
     IPAddress apIP(192, 168, 1, 1);
     ESP8266WebServer webServer(80);
     DNSServer dnsServer;

     const char* ssid = "test";

     void handleRoot();
     void streamFile();
     void setBrightness();

     extern "C" {
       #include "user_interface.h"
       void __run_user_rf_pre_init(void) {
         uint8_t mac[] = MACADDR;
         system_phy_set_max_tpw(TXPOWER);
         wifi_set_phy_mode(PHY_MODE_11G);
         wifi_set_macaddr(SOFTAP_IF, &mac[0]);
       }
     }

     void setup() {
       Serial.begin(115200);
       Serial.setDebugOutput(true);
       ArduinoOTA.begin();
       SPIFFS.begin();

       WiFi.mode(WIFI_AP);
       WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
	     WiFi.softAP(ssid); // 	WiFi.softAP(ssid, password);

       dnsServer.start(53, "*", apIP);

       webServer.onNotFound(handleRoot);
       webServer.on("/", handleRoot);
     webServer.begin();
     }
     
     void loop() {
       dnsServer.processNextRequest();
       webServer.handleClient();
       ArduinoOTA.handle();
     }

     void handleRoot() {
     webServer.send(200, "text/html", "<h1>WORKS!</h1>");
     }

ivankravets, 2016-01-13:
The problem is linked with PlatformIO INO->CPP converter. Temporary solution is to use native `.cpp` instead of `.ino` and place `#include <Arduino.h>` at the top of file.

crushedice2000, 2016-01-13:
@ivankravets: Solved! Thank you very much!

Also is better to use standard cpp than ino or pde!

I've renamed the file to main.cpp :smile_cat:

closed 2016-01-13 by crushedice2000