platformio/platformio-core #458: Streamline SPIFFS upload

aperepel, 2016-01-13:
Hi, I wonder if there is a usable way to optimize the uploadfs target. For esp12e it generates a 3MB file regardless of the data size (usually it's a small json config file and not much more, so mostly padded space).

E.g. guys at Smartjs.io have a good approach of exposing SPIFFS via a WebDAV and being able to add/remove files interactively. I'm not advocating adding this layer, but rather thinking of ways to create a smaller FS image if we know it's gonna be small. And save 5 minutes of waiting for upload to finish.

aperepel, 2016-01-13:
I understand the custom FS images somewhat. The issue is more about defaults. 3MB takes a lot of time and is somehow a default, bad user experience. Add a few more examples of command line invocation in the docs?

ivankravets, 2016-01-13:
How about to use smaller SPIFFS size? http://docs.platformio.org/en/latest/platforms/espressif.html#custom-flash-size

aperepel, 2016-01-13:
Yes, but I don't like our current defaults:
* Use 1MB fs by default - I have hard time imagining why anyone would start with 3MB images and wait 6 minutes on every upload (1MB takes 2 minutes)
* An example from http://docs.platformio.org/en/latest/platforms/espressif.html#custom-flash-size is not very useful as it specifies a 3MB image, which is already a default, so it doesn't change anything. Update this example to use a 1MB image to be more useful.

I guess due to the design of esp12e we only have 1MB or 3MB options, but I'd be happy to be proven wrong.

aperepel, 2016-01-13:
BTW, for some reason, 1MB option causes my file reading code to fail - it can't find the file? The only change was to specify this in `platformio.ini':
```
build_flags=-Wl,-Tesp8266.flash.4m1m.ld
```

ivankravets, 2016-01-13:
Please try http://docs.platformio.org/en/latest/installation.html#d-development-version

aperepel, 2016-01-13:
Ivan, I noticed that switching the FS image size requires one to run a `clean` target, otherwise it gets corrupted or something. Not sure if this can be detected and forced from a config currently, but maybe we can add it to the docs as a best practice/warning? I wasted lots of time debugging that.

Will test out the dev build by tomorrow and report.

ivankravets, 2016-01-13:
If you edit `platformio.ini`, PlatformIO automatically cleanups `.pioenvs` directory. Also, `uploadsf` should force to rebuild SPIFFS image. See https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/espressif.py#L299

Looks, like this is another problem. I'll wait any news from you as for this issue.

ivankravets, 2016-01-13:
I reopen this issue until you test it.

aperepel, 2016-01-14:
Thank you. I have tested the latest dev version of the platform. The defaults are now 1MB for a SPIFF, and FS image rebuild works correctly when going between sizes (tested cycling it up and down several times).

closed 2016-01-14 by aperepel

ivankravets, 2016-01-14:
@aperepel thanks for the report!

commit 88fe8f21 (2016-01-13): Set 1Mb SPIFFS for Espressif boards by default // Resolve #458