platformio/platformio-core #461: Allow library.json to specify sources other than PlatformIO's Repository

dave-newson, 2016-01-16:
### The Problem

If I want to use a `library.json` to manage my projects 3rd party dependencies, then these 3rd party libraries;
- **must** already have a library.json file 
- **must** be registered on PlatformIOs repository.

### Desired Functionality

Let's assume that, in a perfect world:
 - my `library.json` manages **all** the third party dependencies in my project.
 - after I run `platformio update` all dependencies are fulfilled.

The desire is to eliminate the need to manual checkouts, clones, code downloads, etc. In 90% of common use cases I want to just run `platformio update` and have all the magic taken care of.

### Use Case 1 - I want to use a library which isn't on the PlatformIO repository
There are always going to be cases where I want to include a library in my code but the libraries author has not included a `library.json` file in their source.

This may be because:
 - they don't known about PlatformIO
 - they haven't gotten around to putting it in the registry
 - they have some objection to using PlatformIO at all

Frankly, and this is the important bit; **Who cares?** I should not require their consent to perform what is basically `git clone`. The library.json file is a helper for PlatformIOs repository, and not necessary for code to function.

The official solution to this problem is *"submit a library.json PR to the 3rd party"*, but *come on*. That could take weeks to get actioned, and in the mean time I have to manually include the dep. That's hardly a solution.

### Use Case 2 - I want to use a specific fork
Say I've followed the standard GitHub approach of forking, branching, creating a change and then PRing it back to the main third party project. That's nice, and I'm a good open-source citizen.

Problem is, this change won't be available in the library via PlatformIO until it gets merged back in, **if** it gets merged back in.

I could submit my fork up to PlatformIO's repository, so now we have *"arduino-stepper-dave-newson"*, but frankly that's just silly.  Do we want people to do that for every hotfix and branch? No, I don't believe we do.

### Use Case 3: Source code isn't on VCS at all.
There's also cases where libraries have been put online as .zip files by authors born in the 50s and who still write their documentation in .txt files and put things on .edu domains. These libraries simply may never end up on a VCS like GitHub (licensing, anyone?).

The only way around this right now is a manual install.

### Desired Change
*This approach is stolen directly from the PHP package manager Composer.*

I can specify required dependencies in PlatformIO's `library.json` as follows:
```json
"dependencies":
[
    {
        "name": "derp",
    },
```

Please also allow me to override the libraries source with something akin to:
```json
"repositories": [
    {
        "name": "derp",
        "url": "https://github.com/dave-newson/derp.git",
        "type": "git"
    }
],
```

The `name` property indicates to the PlatformIO executable that I want the dependency `derp` to come from the specified GitHub repository. This way I can use forks and branches of my own.

In additional, PlatformIO shouldn't require this repository to contain a `library.json` file. It should just checkout the entire repo wholesale and dump it into my project.

Again, stolen from the likes of Composer, a similar approach can be used to drag in zip archives:
```json
"repositories": [
    {
        "type": "package",
        "package": {
            "name": "derp",
            "version": "master",
            "dist": {
                "type": "zip",
                "url": "http://example.com/src/derp.zip",
                "reference": "master"
            },
        }
    }
]
```

Obviously PlatformIO requires its own approach and json structure, but this would enable people to include from any git-based repository or zip archive, without the need for registration with PlatformIO. 
This would greatly expand its capabilities and enable faster adoption.

This system could also allow expansion for compatibility with any other VCS like Mercurial or SVN.

oskargargas, 2016-01-19:
On top of that please allow checkout of specific version based on git (or other vcs) tag or commit hash.
Ruby's Bundler and iOS Cocoapods (which uses the same library file format) are perfect example whats needed in terms of dependency version management and custom sources checkout.

ivankravets, 2016-01-19:
@oskargargas Of course, I'll implement it.

furyfire, 2016-01-26:
The design specs for PHP's composer should apply directly to platformio.
https://getcomposer.org/doc/
Maybe it is time for the library's you require to be placed in the project folder and not platformio's user folder?

What about resolving dependecies? Can package A require package B which will also be automatically installed and updated?

ivankravets, 2016-01-26:
@furyfire 

> Maybe it is time for the library's you require to be placed in the project folder and not platformio's user folder?

Thanks! See https://github.com/platformio/platformio/issues/475

> What about resolving dependecies? Can package A require package B which will also be automatically installed and updated?

These features are partially implemented in the PlatformIO 2.0. However, this functionality will be rewritten from the scratch in PlatformIO 3.0. See https://github.com/platformio/platformio/issues?utf8=✓&q=is%3Aopen+milestone%3A3.0.0+label%3Alib

closed 2016-08-08 by ivankravets

ivankravets, 2018-09-15:
@danielschenk 

> ""frameworks": "native","

http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-compat-mode

danielschenk, 2018-09-15:
@ivankravets 

When I remove that line, LDF still does not find it 😢. And I actually want to use the compatibility feature of the LDF, because I have different environments in my `platformio.ini`:
- One which builds a little test app for my MIDI logic, which uses an USB MIDI interface and runs on the PC, therefore uses `platform = native` and needs rtmidi.
- One which builds the application for the real target (ESP32 with Arduino framework), using the same MIDI classes but different MIDI driver (using Arduino HardwareSerial).

ivankravets, 2018-09-16:
What does mean framework=native in library.json? Do you mean platform=native?

Anyway, could you provide a simple project to reproduce this issue?

danielschenk, 2018-09-16:
You are right, it should have been `platforms: "native",` in the manifest. I tried it, unfortunately does not make a difference.

I attached a simple project to reproduce the issue. When I run it, compilation fails because including `RtMidi.h` fails.
[platformio-issue-461.zip](https://github.com/platformio/platformio-core/files/2387025/platformio-issue-461.zip)

ivankravets, 2018-09-17:
@danielschenk how can LDF find a library if there is no #include <RtMidi.h>?

danielschenk, 2018-09-17:
@ivankravets The include statement is there, in `Application.cpp`.

If you're talking about the actual header file `RtMidi.h`, you are correct, it's not there. That's because I expect that the library automatically gets downloaded from GitHub by PlatformIO (without first registering it in the database). I was under the impression that @dave-newson wanted the same, and opened this issue for it. The issue got closed, so I thought the idea had been implemented.

If it is not possible what I want, then we unfortunately misunderstood each other.

danielschenk, 2018-09-19:
The source file is already there, in the example I uploaded. It's `src/Application.cpp`. And it includes `RtMidi.h` like said before.

I can imagine that LDF doesn't know that it needs to download RtMidi from GitHub because it doesn't know that `RtMidi.h` (which should be found as dependency by LDF) is actually part of that repository. But maybe a hint could be added to `platform.json` to make this possible? Something like:
```json
{
    "name": "rtmidi",
    "repository": {
        "type": "git",
        "url": "https://github.com/thestk/rtmidi.git",
        "branch": "v3.0.0"
    },
    "frameworks": "native",
    "build": {
        "extraScript": "rtmidi.py"
    },
    "headers": [
        "RtMidi.h"
    ]
}
```

danielschenk, 2018-09-19:
Thanks, I could not find option 2 in the documentation. I will try it out later today :)

Will it still make use of my local `library.json`? Because I need to run the extra script for the RtMidi build flags.

ivankravets, 2018-09-19:
Yes, it will use `library.json` it the last one included in a library. 

See http://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-deps

danielschenk, 2018-09-21:
That's still not exactly what I want. I want to use the original RtMidi from GitHub, but I want `library.json` and the extra script to be in my own project (because I see no point in forking RtMidi just for that).

I attached a new example to make this clear. When I run it, it tries to build a library directly (with no files) and therefore stops with an error. Instead, I expect it to:
1. Understand that it needs to clone RtMidi from GitHub first
1. Then build it with the flags as specified by `rtmidi.py`.

[platformio-issue-461-example2.zip](https://github.com/platformio/platformio-core/files/2406594/platformio-issue-461-example2.zip)

ivankravets, 2018-09-23:
LDF depends on `library.json` manifest. We don't do 3rd party work. If manifest exists, we use this library and don't check source files. There a lot of ways how to resolve your issue:

1. Fork rtmidi librart and update with own `library.json` and extra script. Later, use direct GIT link to your repo
2. Add "lib_deps = https://github.com/thestk/rtmidi.git", PIO will clone this repo, then use `PRE:SCRIPT` and put into the library 2 files: library.json and your extra scripts per library

commit 758396c9 (2016-08-02): PlatformIO Library Manager 3.0 // Resolve #588 , Resolve #414, Resolve #498, Resolve #475, Resolve #410, Resolve #461, Resolve #361, Resolve #507, Resolve #554