platformio/platformio-core #462: Split C/C++ build flags

oskargargas, 2016-01-16:
There are some build flags which are valid only for C files or only for C++ files, ie. `build_flags = -felide-constructors` generates warnings when compiling C files as it is C++ only flag.

ivankravets, 2016-01-18:
This is valid behaviour of general `build_flags` option. If you need more advanced build flags configuration please take a look at http://docs.platformio.org/en/latest/projectconf.html#extra-script

`extra_script.py`

```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()
env.Append(CCFLAGS=["flag1", "fag2"])  # General options that are passed to the C and C++ compilers.
env.Append(CFLAGS=["flag1", "fag2"])  # General options that are passed to the C compiler (C only; not C++).
env.Append(CXXFLAGS=["flag1", "fag2"])  # General options that are passed to the C++ compiler. 
```

More details http://www.scons.org/doc/production/HTML/scons-user.html#cv-CCFLAGS

oskargargas, 2016-01-18:
Thanks, it looks I underestimated `extra_script` capabilities. I'd give it a try.

ivankravets, 2016-01-19:
Happy coding!

closed 2016-01-19 by ivankravets

oskargargas, 2016-01-31:
I'll leave comment here to not open a new issue. Hopefully you'll see this.

Is extra_script executed before or after board builder scripts? I work with Teensy 3.2 and I'm under impression it is executed before though I'd expect it to execute after. In exampleI'd like to override CPPDEFINES set in Teensy builder script https://github.com/platformio/platformio/blob/7a7fa845fe82404b4adff6e598efe6690a20ea55/platformio/builder/scripts/teensy.py#L39. Is it possible in any way besides forking whole PIO?

ivankravets, 2016-01-31:
@oskargargas 

> Is extra_script executed before or after board builder scripts? I work with Teensy 3.2 and I'm under impression it is executed before

After. See https://github.com/platformio/platformio/blob/develop/platformio/builder/main.py#L147

>  In exampleI'd like to override CPPDEFINES 

You can "undefine" any macros using `-U` in `build_flags`.

> Is it possible in any way besides forking whole PIO?

PlatformIO is open source project. You can fork it. See button above.

oskargargas, 2016-01-31:
@ivankravets `-U` works as expected during compilation but doesn't when generating CMake files for CLion what is understandable. 

If I prepare pull request exposing some flag configuration for Teensy board will it be accepted?
I'm thinking about `TEENSY_USB_TYPE` defaulting to `USB_SERIAL`, `TEENSY_KEYBOARD_LAYOUT` defaulting to `LAYOUT_US_ENGLISH` and `CPP_STANDARD` defaulting to `gnu++0x`. All defaults are currently in use.

ivankravets, 2016-01-31:
@oskargargas I accepted 90 PRs for PlatformIO 2.0 see https://github.com/platformio/platformio/pulls?q=is%3Apr+is%3Aclosed

However, PlatformIO 3.0 will be licensed under GPL 2.0. All contributions to the PlatformIO Core (not to scripts, docs) will be removed. I decided to untie contributors hands and create this issue https://github.com/platformio/platformio/issues/479

It means, that code of board presets, platform and framework build scripts will be moved to separated repos and licensed under  current Apache 2.0 license. All of us will have ability to create own platforms or propose new changes to existing.

I received 0 PR to PlatformIO core for 1.5 year. I hoped that people will help me with it but as practice has shown most like only to use. 

I don't have any objections to PlatformIO users and respect all of us. PlatformIO Teams spends our spare time totally for free. We  provide HIGH quality support within 1 hours  (GitHub, E-mails, Gitter) because we love people and what we do for them. 

I'll write separate letter for all PlatformIO users next week and ask them about further fate of PlatformIO. I have a lot of unique ideas for IoT world and PlatformIO team is ready to implement them using your support.

----

P.S: As you for your PR, please wait for https://github.com/platformio/platformio/issues/479 or place here patch and I'll will apply it from own account (because I don't see logic to add some then delete).