platformio/platformio-core #463: Discover OTA-able esp8266/Arduino modules using dns-sd

probonopd, 2016-01-17:
To use OTA on the esp8266, we currently have to put something like `upload_port=192.168.0.16` into `platformio.ini`.

This seems overly static to me. Wouldn't it be great if platformio would detect OTA-able devices just as seamless as it does auto-detect the serial port?

Turns out this would be pretty easy to implement using avahi (there are python bindings!)

```
# Look for suitable devices to upload to
avahi-browse -t -r  "_arduino._tcp"

# Alternatively, just get the IP address of a known hostname
avahi-resolve -h -n esp8266-c1ab73.local

# If there is only one device in the network you can just do:
IP=$(avahi-browse -t -v -r  "_arduino._tcp" 2>/dev/null | grep address | head -n 1 | cut -d "[" -f 2 | cut -d "]" -f 1)
echo $IP
echo "upload_port=$IP" >> platformio.ini
```
We could also check the `TXT` record to make sure it is the correct device model; and we could ask the user to select one from all available update-able devices on the network.

jbbr, 2016-04-11:
In general I like the idea - but I think its important to handle this different from the current serial implementation: Usually one or a few development devices are connected via serial and it's easy to make sure that the only the device(s) you want to flash is(are) connected.
But for OTA: There might be many different devices connected to the network with OTA enabled. How could PlatformIO decide automatically which one to use?

I would suggest to let the user decide by giving a prompt in the terminal.
Additional a commandline parameter to set the upload port/OTA device would be nice (Probably there is already one, haven't checked).

Testato, 2016-12-17:
On arduino ide whit esp core installed, the mDNS discovery is already present.
All OTAble esp is automatically listed under Ide upload port, but is always the user that must set manually the port, because if you have 10 esp in your house the system cannot auto-decide witch esp you want upload.
So i think that Platformio need only do the same arduino behaviour, by create an autodiscovery upload port presence, but not an auto-setting upload port.

This is thrue also for standard serial port, because you can have many esp connected to the same pc, so in the arduino ide you auto-found all available serial port, but after you must manually choose whitch ones you need use

ivankravets, 2016-12-17:
@Testato PlatformIO also allows to manually specify upload port as a serial device, media disk or IP/mDNS record. See

* http://docs.platformio.org/en/latest/projectconf.html#upload-port

Testato, 2016-12-17:
Yes, i know,
but there is no automatic Discovery like in the Official arduino Ide.
You need know witch port use, instead on Arduino IDE the port menu is automatic populated.
So you do not must remember all your mcu.

For example in your home you have 10 ESP with the OTA, every ESP have a unique mDNS name, so on arduino you receive automatically all 10 names in the list, and you will choose to witch upload.

Instead on Platformio you need remember exactly the name of all ESP that you have in your Home

BrandonLWhite, 2017-10-16:
Seeing as how platformio.ini should be revision controlled, I want to avoid putting a line in there that pertains only to my own personal development environment.

ivankravets, 2017-12-18:
Will be reflected in PIO Home -> Devices soon.

commit 31814b51 (2017-12-18): Extend "pio device list" command with new options to list logical devices and multicast DNS services // Resolve #463
closed 2017-12-18 by ivankravets