platformio/platformio-core #464: Make local installs the default

boneskull, 2016-01-18:
Having worked in python for 8 years alongside JavaScript, it's always flummoxed me why the hell pip wants to install packages to a global location by default. I think it also flummoxed izs and the rest of npm team as well. 

in pip's case this is usually in /usr somewhere and it wants superuser privs...to untar some files into PYTHONPATH.  

PlatformIO takes the high road and stuffs everything into the user's home directory, but it is unclear to me why that even happens. 

Why not install libs into a subdir of the current working directory, by default? Don't make me overwrite that setting in all my projects. I swear I will do it!!

ivankravets, 2016-01-18:
> in pip's case this is usually in /usr somewhere and it wants superuser privs...to untar some files into PYTHONPATH.

How about http://stackoverflow.com/questions/7143077/how-can-i-install-packages-in-my-home-folder-with-pip ?

> Why not install libs into a subdir of the current working directory, by default? Don't make me overwrite that setting in all my projects. I swear I will do it!!

Because I don't see logic here to duplicate toolchains, tools, compilers, frameworks == ~2Gb for each project. PlatformIO packages are globally intended for the all PlatformIO-based project.

P.S: I'm Web-dev in the past and know how does work `npm/bower` which you mention each time. We can compare that package manager with `platformio lib`, but not with whole PlatformIO ecosystem.

Please take a look at [PlatformIO 3.0](https://github.com/platformio/platformio/issues/423). PlatformIO 3.0 Library Manager will be more similar to npm/bower/compose.

ivankravets, 2016-07-16:
You can install PlatformIO CLI to local user's home directory. Please use `pip install --user platformio`. 

What is your OS?

ivankravets, 2016-08-25:
Please reopen if you need a help

closed 2016-08-25 by ivankravets