platformio/platformio-core #466: library.json include seems broken

JerrySievert, 2016-01-19:
hello,

i am building a library that i plan on making available via platformio,  i have noticed that no matter what `include` is set to, it always uses whatever is in src instead:

```js
{
  "name": "emdb",
  "description": "A Tiny Embedded Database with Query Capability",
  "keywords": "database, query",
  "repository": {
    "type": "git",
    "url": "https://github.com/celcius-labs/emdb.git"
  },
  "include": "build",
  "frameworks": "*",
  "platforms": "*"
}
```

this does not use the build directory, but instead uses src.

```js
{
  "name": "emdb",
  "description": "A Tiny Embedded Database with Query Capability",
  "keywords": "database, query",
  "repository": {
    "type": "git",
    "url": "https://github.com/celcius-labs/emdb.git"
  },
  "include": [
    "src/*.[ch]",
    "jsmn/*.[ch]"
  ],
  "frameworks": "*",
  "platforms": "*"
}
```

also appears to only use what is in the src directory.  am i doing something wrong here?

thanks!

ivankravets, 2016-01-19:
See https://github.com/celcius-labs/emdb/pull/1

>  i have noticed that no matter what include is set to

`include` here doesn't mean C Preprocessor `#include`. See docs http://docs.platformio.org/en/latest/librarymanager/config.html

> PlatformIO Library Crawler uses `library.json` manifest to extract source code from developer’s location and keeps cleaned library in own Library Storage.

it means that developer explains PlatformIO Crawlers which files should be treated as "source code" and should be built.

As for own `CPPATH` and other flags see https://github.com/platformio/platformio/issues/289

JerrySievert, 2016-01-19:
@ivankravets i understand the intended use of `include`, but i am reporting that it does not appear to be working as intended.  i had all of the source code in `build`, but it still only copied and built what was in `src`, instead of following the directive.

this appears to be a bug, given that case.

ivankravets, 2016-01-19:
Could you merge my PR?

JerrySievert, 2016-01-19:
i have merged your PR - what allows for this to work now is that `jsmn` has been published on platformio, which allows for me to use it as a dependency, but i still believe it is a bug, as it does not seem to correctly copy and build if `include` is set, and there is a `src` directory.

ivankravets, 2016-01-19:
```
(develop)ikravets@innomac /V/S/P/G/p/origin> platformio lib install 569
Installing library [ 569 ]:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The library #569 'emdb' has been successfully installed!
Installing dependencies:
Installing library [ 568 ]:
Already installed
```
```
tree -L 2 /Users/ikravets/.platformio/lib/emdb_ID569
/Users/ikravets/.platformio/lib/emdb_ID569
├── CHANGELOG.md
├── LICENSE
├── Makefile
├── QUERY.md
├── README.md
├── config.mk
├── examples
│   ├── basic
│   └── scan
├── library.json
└── src
    ├── emdb.c
    ├── emdb.h
    ├── emdb_config.h
    ├── json.c
    ├── json.h
    ├── query.c
    ├── query.h
    └── storage
```

JerrySievert, 2016-01-19:
yes, i know that it is working when i add a dependency - let me create an example for you that is not published and does not include dependencies to show you the issue.

ivankravets, 2016-01-19:
@JerrySievert You are right. I see this error
```
In file included from .pioenvs/uno/emdb_ID569/json.c:1:0:
.pioenvs/uno/emdb_ID569/json.h:4:18: fatal error: jsmn.h: No such file or directory
#include "jsmn.h"
^
compilation terminated.
avr-gcc -o .pioenvs/uno/emdb_ID569/query.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/emdb_ID569 .pioenvs/uno/emdb_ID569/query.c
scons: *** [.pioenvs/uno/emdb_ID569/json.o] Error 1
In file included from .pioenvs/uno/emdb_ID569/query.h:4:0,
from .pioenvs/uno/emdb_ID569/query.c:4:
.pioenvs/uno/emdb_ID569/json.h:4:18: fatal error: jsmn.h: No such file or directory
#include "jsmn.h"
^
compilation terminated.
scons: *** [.pioenvs/uno/emdb_ID569/query.o] Error 1
```

I will look on it. Thanks.

JerrySievert, 2016-01-19:
thanks! just finished my test case.  i can add it to github if you want, or can let you run with it.

ivankravets, 2016-01-19:
Do you mean this https://github.com/platformio/platformio/issues/466#issuecomment-172998890 ?

JerrySievert, 2016-01-19:
@ivankravets yes

ivankravets, 2016-01-19:
I've just tried to resolve this issue. This isn't a bug. PlatformIO 2.0 looks for dependent libraries which are included from the source files of project. It doesn't look for nested lib by lib. If you want to enable this feature please see http://docs.platformio.org/en/latest/projectconf.html#lib-dfcyclic

However, PlatformIO 3.0 will analyze `library.json` from the library before build . The temporary solution just `#include <jsmn.h>` in the one of `src` files from project.

ivankravets, 2016-01-24:
Please reopen this issue if you need any help

closed 2016-01-24 by ivankravets