platformio/platformio-core #468: platformio for esp8266  need update

krissfr, 2016-01-21:
Hi,

Since the wifiManager lib has been updated ( https://github.com/tzapu/WiFiManager ), "platformio ci" don't compile my program. My program compil fine with the Arduino IDE and the updated version of https://github.com/esp8266/Arduino and https://github.com/tzapu/WiFiManager .

Can you please update the "platformio ci" for esp8266 with the last version of https://github.com/esp8266/Arduino ?

In the Troubleshooting part of the readme (https://github.com/tzapu/WiFiManager) it say : 
If you get compilation errors, more often than not, you may need to install a newer version of the ESP8266 core for Arduino.

Thank you.

krissfr, 2016-01-21:
i add a part of the compilation error : 

/spiffs_api.o .pioenvs/esp12e/FrameworkArduino/spiffs_hal.o .pioenvs/esp12e/FrameworkArduino/time.o
xtensa-lx106-elf-ranlib .pioenvs/esp12e/libFrameworkArduino.a
.pioenvs/esp12e/WiFiManager-master/WiFiManager.cpp: In member function 'void WiFiManager::handleWifiSave()':
.pioenvs/esp12e/WiFiManager-master/WiFiManager.cpp:462:20: error: 'class IPAddress' has no member named 'fromString'
_sta_static_ip.fromString(server->arg("ip"));
^
.pioenvs/esp12e/WiFiManager-master/WiFiManager.cpp:467:20: error: 'class IPAddress' has no member named 'fromString'
_sta_static_gw.fromString(server->arg("gw"));
^
.pioenvs/esp12e/WiFiManager-master/WiFiManager.cpp:472:20: error: 'class IPAddress' has no member named 'fromString'
_sta_static_sn.fromString(server->arg("sn"));
^
scons: *** [.pioenvs/esp12e/WiFiManager-master/WiFiManager.o] Error 1
========================= [ ERROR ] Took 29.33 seconds =========================

ivankravets, 2016-01-24:
I've just updated library and it was switched to stable version. Please
```
platformio lib uninstall 567
platformio lib install 567
```

krissfr, 2016-01-25:
@ivankravets hi, thank you for your help.

Are you change availaible for the "platformio ci" command too ?
i build my project on shippable with "platfomio ci" command, and i still have the issue.

ivankravets, 2016-01-25:
@krissfr I've not change something. Could you point me to build log?

krissfr, 2016-01-25:
@ivankravets this is a pastbin of my log : http://pastebin.com/fWcfT3rL

ivankravets, 2016-01-25:
Here are 2 problems:

1. PlatformIO uses the latest stable Arduino for ESP8266
2. You use the latest DEVELOP version of dependent libs.

## Solution

Install all libraries from PlatformIO Library Registry using `platformio lib install`. If lib isn't in our registry, we will add it.

krissfr, 2016-01-25:
@ivankravets thank you for your help.
i successfully build with platformio lib install 567 and platformio lib install 89 for WifiManager and PubSubClient libraries. The one that i can't find in the platformio library registry is https://github.com/milesburton/Arduino-Temperature-Control-Library . There is a similar library "Dallas Temperature" (id 54), but not available for esp8266, so for this one a continue to download the library from the github repository.

Thank you again :)

ivankravets, 2016-01-25:
> There is a similar library "Dallas Temperature" (id 54), but not available for esp8266

Just installed it :) This only compatibility info. If it works for you, we will make PR to this lib with the new compatible platforms.

ivankravets, 2016-01-25:
I've just made PR https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/50

krissfr, 2016-01-25:
@ivankravets you're right the library id 54 work fine with my esp8266.
Thank you again for your help, now i don't use external libraries, embedded platformio libraries from registry are excellent and easy to use.

closed 2016-01-25 by krissfr

ivankravets, 2016-01-25:
Happy coding! :+1:

milesburton, 2016-01-25:
The TCL works fine with ESP8266 - I'm using it to monitor a couple of Fish Tanks :). Let me review the PR

krissfr, 2016-01-25:
@milesburton i use it for  my reeftank too, to monitor my high power leds temperature ( the esp8266 control leds (sunrise, sunset, cloud, storm, moon, ...), monitor temperature of leds,  control fans to cool Leds, monitor rpm of fans, ...)
@ivankravets and @milesburton thanks to you (and many others who provides libraries), it come true and my corals and fishes are happy :)