platformio/platformio-core #469: SPIFFS upload issue

comino, 2016-01-21:
Platformio 2.7.1
Board: ESP12e
Framework: Arduino 

```
$ platformio run -t uploadfs
[01/21/16 22:53:45] Processing esp12e (platform: espressif, upload_port: 192.168                                        .178.74, board: esp12e, framework: arduino, build_flags: -Wl,-Tesp8266.flash.4m.                                        ld)

"C:\Users\XXX\.platformio\packages\tool-mkspiffs\mkspiffs" -c data -p 256 -b 8192 -s 3125248 .pioenvs\esp12e\spiffs.bin
warning: can't read source directory
scons: *** [.pioenvs\esp12e\spiffs.bin] Error 1
 [ ERROR ] Took 2.74 seconds

```
SPIFFS.bin is succesfully created, but not flashed. 

As also observed by FWeinb: https://github.com/platformio/platformio/issues/382

ivankravets, 2016-01-21:
Do you have files in `data` dir?
Why the print-log is broken?

comino, 2016-01-21:
Oh deam - it was a typo in the dir put in platformio.ini

closed 2016-01-21 by comino

inter-coder, 2018-03-14:
I have same problem but no typo

Wed Mar 14 09:03:49 2018] Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
-------------------------------------
Verbose mode can be enabled via `-v,--verbose` option
PLATFORM: Espressif 8266 > EspressifESP8266 ESP-12E
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting IRServerRes.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebSockets> v2.1.0
|   |-- <SPI> v1.0
|   |-- <Hash> v1.0
|   |-- <ESP8266WiFi> v1.0
|   |-- <Ethernet(esp8266)> v1.0.4
|   |   |-- <SPI> v1.0
|-- <IRremoteESP8266> v2.3.3
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
Building SPIFFS image from 'data' directory to .pioenvs/esp12e/spiffs.bin
warning: can't read source directory
*** [.pioenvs/esp12e/spiffs.bin] Error 1
===== [ERROR] Took 1.06 seconds =====
The terminal process terminated withexit code: 1

Terminal will be reused by tasks, press any key to close it.