platformio/platformio-core #471: Add support for Infineon XMC1000/XMC1100 (XMC 2GO) and XMC4000/XMC4500 (XMC 4500 Relax) Familiy

nmaas87, 2016-01-24:
http://www.infineon.com/cms/en/product/microcontroller/32-bit-industrial-microcontroller-based-on-arm-registered-cortex-registered-m/xmc-development-tools-kits-and-boards/channel.html?channel=db3a30433580b3710135a07902883872

http://www.infineon.com/cms/en/product/evaluation-boards/KIT_XMC_2GO_XMC1100_V1/productType.html?productType=db3a304443537c4e01436ccecb5d154f#ispnTab1

http://www.infineon.com/cms/en/product/evaluation-boards/KIT_XMC45_RELAX_V1/productType.html?productType=db3a304437849205013813b23ac17763

rezasurmar, 2017-07-19:
Any progres? Is it possible to import from Arduino IDE?

ivankravets, 2017-07-19:
@rezasurmar could you provide links to Arduino IDE and Infineon XMC?

rezasurmar, 2017-07-19:
https://github.com/Infineon/XMC-for-Arduino/wiki/Implementation-in-Arduino-IDE

PabloPL, 2018-03-15:
Any update on this? It would be nice to have this supported in PlatformIO (so we don't have to use Arduino IDE).

sherylll, 2018-06-15:
https://github.com/Infineon/platformio-infineonxmc

ivankravets, 2018-07-19:
- https://platformio.org/platforms/infineonxmc

Please open PlatformIO IDE > PIO Home > Platforms

commit e4462d75 (2018-07-19): Add Infineon XMC dev/platform // Resolve #471
closed 2018-07-19 by ivankravets