platformio/platformio-core #472: Add please Arduino Boards from arduino.org

sstaub, 2016-01-25:
Please can you add the boards from the alternativ Arduino Company arduino.org
- Leonardo ETH
- M0
- M0 Pro

sstaub

choidamdam, 2016-02-03:
+1
maybe we should create separate framework like `arduino.org`.

ivankravets, 2016-02-03:
I contacted with Arduino.Org and waiting for the further information. I'll write here about all news.

choidamdam, 2016-03-17:
There are some new arduino.org-only boards:
- [Arduino Tian](http://www.arduino.org/products/boards/4-arduino-boards/arduino-tian)
- [Arduino Industrial 101](http://www.arduino.org/products/boards/4-arduino-boards/arduino-industrial-101)
- [Arduino Yun Mini](http://www.arduino.org/products/boards/4-arduino-boards/arduino-yun-mini)

aethaniel, 2016-03-23:
@valeros, this ticket should not be labelled vendor:atmel :wink:

ivankravets, 2016-03-23:
@aethaniel thanks :+1: I've just removed :blush:

ivankravets, 2016-04-28:
PlatformIO 2.9.0: 
* Added support for Arduino.Org boards: Arduino Leonardo ETH, Arduino Yun Mini, Arduino Industrial 101 and Linino One

hgijeon, 2016-05-13:
I couldn't find neither 'M0' nor 'M0 pro' in boards list.
Could you add those boards?

ivankravets, 2016-06-24:
@sstaub @choidamdam @hgijeon ,

Do you have these boards?

* Arduino Tian
* M0
* M0 Pro

We would ask you to test this issue. Thanks!

sstaub, 2016-06-24:
Hi Ivan,
I have a M0 Pro board from arduino.org <http://arduino.org/> and I can test it

Regards

Stefan Staub

valeros, 2016-06-24:
Hi @sstaub !
It would be great if you could test it. Please follow next steps:

1. Install PlatformIO from this branch:
`pip install -U https://github.com/platformio/platformio/archive/feature/arduino-org-samd-issue-472.zip`
2. Put [this file](https://dl.dropboxusercontent.com/u/10025728/samd21g18a_bootloader_org.ld) to `ldscripts` package in your PlatformIO home directory:
`/home/user/.platformio/packages/ldscripts`
3. Try uploading to different ports (`mzero_pro_dbg` board for programming port, `mzero_pro` for native port)

Thanks in advance!

sstaub, 2016-06-24:
Hi Valerii,
I installed the items as you told. I hope the version is right (what the cmd to check the installed version?).
The file in 2., I copied the content manually in my text editor (mac), I hope this is ok, maybe you send me
this .ld file as an attachment.
So, I started:
1. There is no board „M0“ or „M0 pro“ only „Zero“ 
2. I tried Zero (Programing Port“
	- I need manually „Rebuild C/C++ Project Index“ because aruino.h is not found
	- building Ok
	- upload -> see screenshot, the board stops working 

3. I tried "Zero (USB Native Port)“
	- I need manually „Rebuild C/C++ Project Index“ because aruino.h is not found,
	after changing the board, the proceed of automatic rebuild doesn’t work
	- building Ok
	- upload -> see screenshot, one led is blinking very fast (so it seems there is an upload) but nothing happened

The upload generally works only with Arduino Studio (alpha 0.5) from arduino.org <http://arduino.org/>, when trying upload with
Arduin 1.6.9 fails because the Arduino IDE can’t find the USB port. There are some special differences
between M0 and Zero, they are not compatible.

Regrads

Stefan Staub

ivankravets, 2016-06-25:
> I installed the items as you told. I hope the version is right (what the cmd to check the installed version?).

Where did you install it? Did you use PlatformIO IDE Terminal?

> The file in 2., I copied the content manually in my text editor (mac), I hope this is ok, maybe you send me
this .ld file as an attachment.

Please right click on [this link](https://dl.dropboxusercontent.com/u/10025728/samd21g18a_bootloader_org.ld) and "Download File..." from a context menu.

> There is no board „M0“ or „M0 pro“ only „Zero“

If you use PlatformIO IDE and installed here this test branch, you need to restart IDE.

sstaub, 2016-06-25:
Hi, I installed it first with terminal and today I make a clean install and
make an update to dev. version with build in terminal.
The .ld file can’t downloaded with right click on mail OS X.
Here is the selfdone copy:

The version on home screen is 2.10.3, i expected 2.10.4dev2 ??? the option for using the dev.
version was set. I unset and reset the option, than the version in homescreen is ok 2.10.4dev2.
It seems it is generally a problem when updating via terminal.
The problems are the same like yesterday, It seems that platformio can not find
the usb device. That happens also when I try to upload with the arduino IDE from
arduino.cc <http://arduino.cc/> ! It seems that uploading will work with the original „Zero“, but I can’t test it
because the my „M0 pro“ comes from arduino.org <http://arduino.org/> !
Maybe you contact the developer team of arduino.org <http://arduino.org/> and find someone who have an „original“
zero board.

stefan

PS.: what is the command to check the version

valeros, 2016-06-27:
Hi @sstaub ! @ivankravets told me that uploading over `avrdude` breaks the board.
Could you please replace default `arduinosam` package `/home/user/.platformio/package/framework-arduinosam` with this [one](https://www.dropbox.com/s/7dk9x7ou3z6vrl5/framework-arduinosam_m0.zip?dl=0) and try uploading for `mzero_pro` board (native port)? Any changes?

sstaub, 2016-06-27:
Hi Valerii,
I replaced the file but there is no more upload on native usb port possible, i get a time out:

stefan

valeros, 2016-06-27:
@sstaub Have you restored the board?
Would be great if you could try to upload firmware from your terminal via `avrdude` shipped with Arduino IDE (1.7.10).
Something like that:
```
/home/user/path_to_arduino_ide/hardware/tools/avr/bin/avrdude -C /home/user/path_to_arduino_ide/hardware/tools/avr/etc/avrdude.conf -v -v -p atmega2560 -c stk500v2 -P /dev/your_device  -b 57600 U flash:w:/home/user/path_to_platformio_project/.pioenvs/env/firmware.hex:i
```
Thanks for your help!

ivankravets, 2016-07-05:
@sstaub @hgijeon,

Please try the latest http://docs.platformio.org/en/latest/installation.html#development-version . Tian, M0 and M0 Pro via USB should work now.

Don't forget to update all packages via `platformio update`.

Please answer here if it works for you.

sstaub, 2016-07-05:
Hi Ivan
i tested it on the M0 pro and the problem is also the same at the last session,
when uploading code to the native usb-port the boot loader doesn’t work any more,
so I cannot upload the next time. To reset, i have to upload with the arduino.org <http://arduino.org/> ide
an usb application over the programming port, then it work for a next try, then it stops again...

Regards

Stefan Staub

ivankravets, 2016-07-11:
@sstaub Could I ask you to re-test the latest `2.11.1b1` build with Arduino M0?

1. PlatformIO IDE `Menu: PlatformIO > Upgrade PlatformIO` (if it is switched to `develop`)
2. PlatformIO IDE `Menu: PlatformIO > Update packages`

Please check in PlatformIO IDE Terminal `pio --version`. It should be `2.11.b1`.

Thanks.

ivankravets, 2016-07-11:
@sstaub,  @choidamdam,  @aethaniel, @hgijeon 

Please re-test with the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

If you have any problems, feel free to reopen this issue.

closed 2016-07-11 by ivankravets

sstaub, 2016-07-20:
Hello,

Here is a good news from our mutual  friend, just read  it here <http://idea.thistastesgood.com/e3plcw>

Rushing, Stefan Staub

sstaub, 2016-07-24:
Dear! 

I've just  read an  article  and I have some questions  that you may really    know the answer, please read  it here <http://official.jansendesil.com/e3ywfvk>

Looking forward, Stefan Staub

49047759