platformio/platformio-core #477: Incorrect board when uploading nodemcu instead of esp12e

eyp, 2016-01-26:
I'm working on a project with board configured _esp12e_, but what is my surprise that when it uploads the firmware via serial port, it sets board _nodecmu_, and my device doesn't work right... 

My platformio.ini file:

[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
build_flags = -Wl,-Tesp8266.flash.4m.ld

The output of the command platformio -t upload

[Tue Jan 26 18:42:32 2016] Processing esp12e (platform: espressif, board: esp12e, framework: arduino, build_flags: -Wl,-Tesp8266.flash.4m.ld)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs/esp12e/firmware.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.usbserial-AL01BY0E
"/Users/edu/.platformio/packages/tool-esptool/esptool" -vv -cd nodemcu -cb 115200 -cp /dev/cu.usbserial-AL01BY0E -cf .pioenvs/esp12e/firmware.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
**setting board to nodemcu**
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.usbserial-AL01BY0E
espcomm_upload_file
stat .pioenvs/esp12e/firmware.bin success
opening port /dev/cu.usbserial-AL01BY0E at 115200

ivankravets, 2016-01-26:
> setting board to nodemcu

This is reset method `-cd nodemcu `. Change it to `ck`, see http://docs.platformio.org/en/latest/platforms/espressif.html#custom-reset-method

P.S: `reset_method` is implemented in PlatformIO 2.8.0 which has not been released yet. You can try the latest [development version](http://docs.platformio.org/en/latest/installation.html#d-development-version) or [PlatformIO IDE](http://docs.platformio.org/en/latest/ide/atom.html) which uses PlatformIO 2.8.0.

eyp, 2016-01-26:
After installing last 2.8.0 dev4 version, I can build the project but can't upload:

Tue Jan 26 19:21:44 2016] Processing esp12e (platform: espressif, upload_resetmethod: ck, board: esp12e, framework: arduino, build_flags: -Wl,-Tesp8266.flash.4m.ld)

<lambda>(["upload"], [".pioenvs/esp12e/firmware.bin"])
scons: *** [upload] TypeError : decoding Unicode is not supported
Traceback (most recent call last):
File "/Users/edu/.platformio/packages/tool-scons/script/../engine/SCons/Action.py", line 1063, in execute
result = self.execfunction(target=target, source=rsources, env=env)
File "/Library/Python/2.7/site-packages/platformio/builder/scripts/espressif.py", line 337, in <lambda>
[lambda target, source, env: env.AutodetectUploadPort(), "$UPLOADCMD"])
File "/Users/edu/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224, in __call__
return self.method(*nargs, **kwargs)
File "/Library/Python/2.7/site-packages/platformio/builder/tools/pioupload.py", line 92, in AutodetectUploadPort
for item in get_serialports():
File "/Library/Python/2.7/site-packages/platformio/util.py", line 277, in get_serialports
for p, d, h in comports() if p]
TypeError: decoding Unicode is not supported

ivankravets, 2016-01-26:
> TypeError: decoding Unicode is not supported

I see it, give me a few minutes it will be fixed.

ivankravets, 2016-01-26:
@eyp please reinstall the latest PlatformIO 2.8.0.dev5

eyp, 2016-01-26:
Done. Well, now it builds and uploads without the nodecmu board (uses ck) but stills the behaviour is different to what I get when building with Arduino IDE. I mean, it doesn't work as I expect, so while I try to see what happens I'll come back to Arduino IDE :(... that what's something I wanted to avoid.

Thanks Ivan.

ivankravets, 2016-01-26:
@eyp which is your settings in Arduino IDE? Reset method? CPU speed? LD Script?

eyp, 2016-01-26:
I've found the reason. If I use esptool it doesn't flash it right, but if I use esptool.py it works. I've tested in Arduino IDE, and with esptool 0.4.6 it doesn't flash it fine. I forgot to say I'm on OSX.

eyp, 2016-01-26:
Anyway, knowing that I can build the project with PlatformIO,  I think I can live uploading using esptool.py on the command line :).

Thanks!

ivankravets, 2016-01-26:
> So, is it possible to use esptool.py with platformio?, or there's a way to tell it to use it instead of esptool?

Sure! Please take a look at [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script). See my example https://github.com/platformio/platformio/issues/365#issuecomment-163695011

`extra_script.py`
```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    UPLOADCMD='/usr/local/bin/esptool.py --port $UPLOAD_PORT write_flash 0x00000 $SOURCE'
)
```

ivankravets, 2016-01-26:
> Anyway, knowing that I can build the project with PlatformIO, I think I can live uploading using esptool.py 

This is not good! You should report your problem to @igrr https://github.com/igrr/esptool-ck/issues

eyp, 2016-01-27:
The problem is that I don't know what happens when I upload with esptool. I mean, my device has a multi color led that shows a different color depending on its state. Also prints some traces to the serial port. Well, when I upload with esptool, it doesn't print anything, and the led behaviour is crazy. But I actually don't what's happening in the chip.

Anyway I'll try to explain my case there.

And thanks so much to you for your time and your tool, at last I can get rid of Arduino IDE :)

closed 2016-01-27 by eyp