platformio/platformio-core #478: Donating to PlatformIO development

jasonmhite, 2016-01-26:
I really like PlatformIO and would like to support its development. I'm just a student so it wouldn't be anything huge, but I'd like to make a donation. Is there a way to set up to do this? I searched here and on the main page and couldn't find any mention apart from Bountysource, which isn't the same thing as donating.

ivankravets, 2016-01-26:
Thanks a lot!!! I've just added button to the README page https://github.com/platformio/platformio

P.S: We use our local Ukrainian payment system because PayPal doesn't work in Ukraine.

ivankravets, 2016-02-17:
See new page http://platformio.org/#!/donate

closed 2016-02-17 by ivankravets

jasonmhite, 2016-02-23:
Ah yes, that is much better. I tried to donate with the old system, but it wouldn't authorize my payment. The new Bountysource donations work much better.

Keep up the good work! I keep hearing more and more people praising PlatformIO, so you definitely have people's attention.

ivankravets, 2016-02-23:
Thanks a lot for the donation and warm words!

commit 62d226c8 (2016-02-17): Update donate link // Resolve #478