platformio/platformio-core #479: "platform.json" manifest for development platform

ivankravets, 2016-01-27:
# Aims

We are very happy that PlatformIO community is growing up extremely fast. The last weeks have brought us a lot of requests to add new board or platform https://github.com/platformio/platformio/issues. This forces us to refactor PlatformIO 3.0 development platforms logic from the scratch. The main aim is to allow users, developers or even embedded vendors to create own "development platform" which can be easily installed using PlatformIO.

# Requirements

## Platform developer

Developer of "development platform" should provide the next information for PlatformIO Platform Manager:

* Generic platform information (name, version, etc)
* Package dependencies
* Platform build script
* Frameworks (supported by this platform) build scripts
* Boards list (only for embedded platforms)

**Package dependencies** should allow to use `semver` notation. For example, `"framework-mbed": "~1.2.0". In this case default PlatformIO Package Storage should be used. If package does not exist in PlatformIO Package Storage or developer wants to use own package, then allow to specify "direct download link" to it.

## PlatformIO Core

### PlatformIO Platforms Registry

Need to implement public Platforms Registry where will be located information about existing "development platforms". It will by used by http://docs.platformio.org/en/latest/userguide/platforms/cmd_search.html

# Using

# Platform is registered in Platforms Registry

It should look from the CLI as generic CLI for PlatformIO 2.0http://docs.platformio.org/en/latest/userguide/platforms/cmd_install.html

# Platform is not registered in Platforms Registry

Allow to install it directly using archive or folder which contains `platform.json`

```
platformio platforms install file:///path/to/platform.zip
platformio platforms install https://github.com/path/to/platform.zip
platformio platforms install file:///path/to/platform/dir
```

Comments are welcome!

beriberikix, 2016-03-15:
Awesome! Would installing from git also be supported? NPM has a good [example for this](https://docs.npmjs.com/files/package.json#git-urls-as-dependencies).

ivankravets, 2016-03-15:
@beriberikix I don't see any problems to not support CVS. However, need to understand that CVS clients are not installed on the all host OS. Maybe, some warnings to user?

closed 2016-06-11 by ivankravets

ivankravets, 2016-06-11:
Examples: http://docs.platformio.org/en/feature-platformio-30/platforms/creating_platform.html#examples

commit 41617f55 (2016-06-11): Update documentation for the new decentralised development platforms // Resolve #479