platformio/platformio-core #480: Adding support for Lattice FPGAs and Icestorm opensource fpga toolchain

Obijuan, 2016-01-28:
In mid 2015, Clifford Wolf released the first opensource toolschain for FPGAs: [The icestorm project](http://www.clifford.at/icestorm/)

It would be great to include it into Platformio. Having an IoT ecosystem that include microprocessors, FPGAs and soft-processors will be amazing.  Platformio can be the first opensource tools to support that whole ecosystem.

We have started the integration. The results are in this github repo/wiki:

https://github.com/bqlabs/Platformio-FPGA/wiki/Platformio-FPGA-wiki-home

We initially have started with the support for the [Lattice icestick boards](http://www.latticesemi.com/icestick). We already have a working "hello world" FPGA example. When the command "platformio run" is executed, the leds.v verilog file is synthetized and the bitstream generated. With "platformio run --target uploaded" it is uploaded into the FPGA.

We need some tips on how is best way to proceed for integration into platformio. We will also need some tips on how to package the icestorm tools so that they can be automatically installed when the fpga project are created. We will do all the work, we just need some advices and documentation :-)

Thanks a lot

Obijuan, 2016-02-01:
We already have the package  toochain-icestorm-linux_x86_64-1.tar.gz, with all the icestorm tools compiled  for Linux_x86_64. How should be proceed? Where should we upload the package?
Thanks a lot

ivankravets, 2016-02-01:
@Obijuan Hi! Is difficult to provide toolchain for other host OS?

> Where should we upload the package?

With PlatformIO 3.0 you can keep it in own location. I'm working on it. However, if you don't wait for PlatformIO 3.0 I can place it to current [PlatformIO Storage](https://sourceforge.net/projects/platformio-storage/). Where I can download it?

Obijuan, 2016-02-01:
Thanks @ivankravets!

> Hi! Is difficult to provide toolchain for other host OS?

No. We will do it. But first we want to develop under linux, to understand all the details of platformio. Once everything is working fine, we will provide packages for the other host OS

> With PlatformIO 3.0 you can keep it in own location. I'm working on it. However, if you don't wait for PlatformIO 3.0 I can place it to current PlatformIO Storage.  

That would be great :-)

> Where I can download it?

[toochain-icestorm-linux_x86_64-1.tar.gz](https://github.com/bqlabs/Platformio-FPGA/raw/master/packages/toochain-icestorm-linux_x86_64-1.tar.gz)

Thanks a lot :-)

Obijuan, 2016-02-02:
I am preparing the pull-request for this issue. 

Should I created a branch of develop? or a branch of feature/platformio-30

Thanks

ivankravets, 2016-02-02:
The package has been added to manifest and mirrored over CDN. See http://api.platformio.org/packages/manifest

Where I can download the packages for the rest archs? 

Thanks! :+1:

Obijuan, 2016-02-03:
Thanks a lot!  I will prepare the packages for the other archs. It will requiere some time, as we only works on Linux. I will try to give them to you ASAP

Obijuan, 2016-02-04:
Hi @ivankravets !

This is the toolchain-icestorm package for x86_64 **Version 2**:  

[toolchain-icestorm-x86_64-2.tar.gz](https://github.com/bqlabs/Platformio-FPGA/raw/master/packages/toolchain-icestorm-x86_64-2.tar.gz)

We have recompiled the tools to minimize the external shared libs. Now almost all the tools have been compiled with static libraries (It fixes some user issues)

ivankravets, 2016-02-21:
@Obijuan the link is broken. Should I update this toolchain?

Obijuan, 2016-02-22:
Hi @ivankravets!

We have been working hard on the FPGA platform. The updated toolchain for the x86_64 arch is:

[toolchain-icestorm-x86_64-5.tar.gz](https://github.com/bqlabs/toolchain-icestorm/releases/download/0.5/toolchain-icestorm-x86_64-5.tar.gz)

It would be great if you could update the toolchain.

We are now working on porting this toolchain to other archs

Thanks a lot :-)

Obijuan, 2016-02-26:
Hi @ivankravets 

We already have a package for the linux-i686 arch:

[toolchain-icestorm-linux_i686-7.tar.gz](https://github.com/bqlabs/toolchain-icestorm/releases/download/0.7/toolchain-icestorm-linux_i686-7.tar.gz)

It would help us a lot if you could upload to the platformio storage.  There are many people eager to use the FPGAs with platformio

Thanks

ivankravets, 2016-02-26:
@valeros please upload both packages to PlatformIO's Storage

Obijuan, 2016-03-02:
Thanks a lot @valeros ! :-)  It is working like a charm!   Now we can proceed to generate the tools for the other platforms: mac and window. I will put here the links to the packages once they are generated and tested

Obijuan, 2016-03-06:
@valeros, @ivankravets :  We already have packages for windows and mac. It would be great if you could upload them to the platformio storage.  Thanks a lot

[toolchain-icestorm-darwin-7.tar.gz](https://github.com/bqlabs/toolchain-icestorm/releases/download/0.7/toolchain-icestorm-darwin-7.tar.gz)

[toolchain-icestorm-windows-7.zip](https://github.com/bqlabs/toolchain-icestorm/releases/download/0.7/toolchain-icestorm-windows-7.zip)

valeros, 2016-03-07:
@Obijuan Uploaded!

Obijuan, 2016-03-14:
Thanks you so much!!  :-)

ivankravets, 2016-03-19:
Dear @Obijuan, 

Sorry for the stall with this FPGA platform. We are very busy with PlatformIO IDE, bug fixing and PlatformIO improvements. We decided to release PlatformIO 2.9 with a few new platforms and improvements because we don't know when PlatformIO 3.0 will be ready.

I see that your code is licensed under "Creative Commons Attribution-ShareAlike 4.0 International License" where PlatformIO is licensed under Apache v2.0. If you don't have any objections to our license we are welcome to make PR. Please note that all boards, platforms, build scripts will be moved to separate repositories in PlatformIO 3.0.

Also, I see in previous PR that you included binaries/firmwares. Please don't make PR with binaries. I recommend you to remove existing fork what do you have and fork current development version. Make appropriate changes and we will merge it!

In any case, thanks a lot for your efforts!

ivankravets, 2016-03-19:
See PlatformIO 2.9.0 https://github.com/platformio/platformio/milestones/2.9.0

Obijuan, 2016-03-20:
Hi @ivankravets !  Thanks a lot for your response!  No problem at all with the license. I will create a new fork and remove the binaries  :-)

Obijuan, 2016-03-21:
Pull request emitted. I've tested it in Linux, windows and Mac
Platformio will become the first opensource IDE in giving support for FPGAs! That's a big milestone! :-)

ivankravets, 2016-04-26:
Sorry for the delay!!!! 😢 

2.9.0 is going to be released.... TODAY! 😸 

Please re-test the latest http://docs.platformio.org/en/latest/installation.html#development-version

closed 2016-04-26 by ivankravets

Obijuan, 2016-04-26:
Wow! Amazing!  I've re-test the fpga support in the development version and... It works!  :-)
Thanks a lot!

ivankravets, 2016-04-26:
Thank you too! 👍 

P.S: I fixed your code for Python 2.6 https://github.com/platformio/platformio/commit/c5d961e27ead1d4d57909a748fc05ea78a021101. Please update in your repo.

ivankravets, 2016-04-26:
@Obijuan How is difficult to preapre toolchain for ARM v6+ ? (Raspberry Pi adn etc)

Obijuan, 2016-04-26:
We already have these packages:

[toolchain-icestorm-linux_armv7l-7.tar.gz](https://github.com/FPGAwars/toolchain-icestorm/releases/download/0.7/toolchain-icestorm-linux_armv7l-7.tar.gz) -> For Raspberry pi II and Ubuntu phone

[toolchain-icestorm-linux_aarch64-7.tar.gz](https://github.com/FPGAwars/toolchain-icestorm/releases/download/0.7/toolchain-icestorm-linux_aarch64-7.tar.gz) --> ARM 64 bits (ex. Ubuntu tablet)

Obijuan, 2016-04-26:
All the packages are being maintained in this repo:

[Toolchain-icestorm](https://github.com/FPGAwars/toolchain-icestorm/wiki)

commit b51d95ab (2016-04-26): New Lattice iCE40 FPGA development platform // Resolve #480