platformio/platformio-core #481: Add support for AVR XMEGA

knutel, 2016-01-28:
It would be great to have support for AVR XMEGA, too. There already is a project to make Arduino support at [https://github.com/Xmegaduino/Xmegaduino](url), so maybe PlatformIO support is not that difficult. I might take a stab at it myself if I get the time.

aethaniel, 2016-03-23:
Hi @knutel, I took a look at Xmegaduino repo and also did some google search around boards based on Xmega.
Unfortunately, it seems the Xmegaduino would need some love to turn into a dedicated Arduino 1.6.x module and also some available boards.
Furthermore, this arduino port doesn't seem to be maintained since years and is still in beta. 

I didn't seen any strong request for Xmega support in Arduino forum either but it can be explained by all small ARM Cortex-M0(+) devices fighting on the market segment.

Besides that, AVR GCC and all needed tools are available for out-of-framework tinkering.

knutel, 2016-03-27:
Agreed, maybe the effort required is better spent elsewhere. Feel free to close this one.

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-atmelavr/issues/21

closed 2016-11-27 by ivankravets