platformio/platformio-core #486: upload fails and wipes bootloader on Trinket 5v 8Mhz

marcpoppleton, 2016-01-29:
Hi,

I am encountering a problem using PlatformIO to upload a firmware on an Adafruit Trinket 5v 8Mhz.
Compilation ends ok, but upload fails and wipes out the bootloader of the Trinket. Using the Arduino IDE to compile and upload works fine.

Here is the log of the upload using PlatformIO:

```
BeforeUpload(["upload"], [".pioenvs/trinket5/firmware.hex"])
"/Users/marcpoppleton/.platformio/packages/tool-avrdude/avrdude" -v -p attiny85 -C "/Users/marcpoppleton/.platformio/packages/tool-avrdude/avrdude.conf" -c usbtiny -D -U flash:w:.pioenvs/trinket5/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/marcpoppleton/.platformio/packages/tool-avrdude/avrdude.conf"
User configuration file is "/Users/marcpoppleton/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 001:016-1781-0c9f-ff-00
AVR Part                      : ATtiny85
Chip Erase delay              : 900000 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : possible i/o
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    12     4    0 no        512    4      0  4000  4500 0xff 0xff
flash         65     6    32    0 yes      8192   64    128 30000 30000 0xff 0xff
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00

Programmer Type : USBtiny
Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ".pioenvs/trinket5/firmware.hex"
avrdude: writing flash (816 bytes):

Writing | ###avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
###avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 4 retries during SPI command
####avrdude: 3 retries during SPI command
####avrdude: 3 retries during SPI command
#### | 100% 0.51s

avrdude: 816 bytes of flash written
avrdude: verifying flash memory against .pioenvs/trinket5/firmware.hex:
avrdude: load data flash data from input file .pioenvs/trinket5/firmware.hex:
avrdude: input file .pioenvs/trinket5/firmware.hex contains 816 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.07s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0040
0x01 != 0xcf
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (H:00, E:00, L:00)

avrdude done.  Thank you.

scons: *** [upload] Error 1
========================================= [ ERROR ] Took 8.36 seconds =========================================
```

Any idea of what I'm doing wrong?

PlatformIO, version 2.7.1.

Thanks,

ivankravets, 2016-01-29:
Have you tried [Upload using Programmer](http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer)?

marcpoppleton, 2016-01-29:
Yeay, problem solved!
Thanks! (for the answer and the very short deplay to provide it!)

ivankravets, 2016-01-29:
Please give us a star :star: https://github.com/platformio/platformio/stargazers

Have you tried [PlatformIO IDE](http://docs.platformio.org/en/latest/ide/atom.html) ?

marcpoppleton, 2016-01-29:
I use IntelliJ IDEA with Ant scripts calling PlatformIO, I'll give PlatformIO IDE a try. (star given)

ivankravets, 2016-01-29:
Thanks a lot! 

Happy coding!

closed 2016-01-29 by ivankravets