platformio/platformio-core #490: URGENT: Further fate of PlatformIO Project...

ivankravets, 2016-02-01:
Dear PlatformIO Users,

We, [PlatformIO team](https://github.com/orgs/platformio/people), want to say you big THANKS A LOT that use PlatformIO, report issues, share ideas or recommend PlatformIO to your friends. Without you will not be PlatformIO because the main aim of PlatformIO is to be independent from vendors and their influence. We have different tools, scripts, utilities, IDEs over Internet but all of them in most cases are focused on the specific vendor. As result, you have different software/IDE for the different hardware. This is why PlatformIO was born...

# Philosophy

[20 months ago](https://github.com/platformio/platformio/commit/e0c4fb36cf24739e8413e6c5acfcf61360cee625) PlatformIO received the first commit under MIT License. The main aim of PlatformIO's founder ( @ivankravets ) was to
* keep PlatformIO independent from business influence
* keep it totally for FREE even for commercial using
* provide HIGH quality support for FREE (e-mails, GitHub issues, chat, remote sessions)
* build team of contributors around PlatformIO which will support it for the further years.

All of these items/rules have worked except the last. Please take a look at [contributors picture](https://github.com/platformio/platformio/graphs/contributors). However, while you have rested or played with your embedded projects, PlatformIO team worked in own SPARE time (weekends, nights). Why? Because each of us has main work... Nevertheless, we don't have any objections that "people like to use and do not want to contribute" because we are from Ukraine and we love people and love to make them happier.

# Success and Failure

We were happy for the first 18 months because:
* we had time to support this project 
* we had time to add new boards, platforms
* we had time to make people happy with PlatformIO.

What has been changed for the last 2 months? These changes from the business model are big **SUCCESS**. Though, for PlatformIO team it's big **FAILURE**.
[image]

# Huge hole...

We have stuck in the huge hole... We received [a lot of issues](https://github.com/platformio/platformio/issues?q=is%3Aopen+is%3Aissue) to add new boards, platforms and implement new features. Also, we have started work on the next generation [PlatformIO IDE](http://platformio.org/#!/get-started) for IoT development. As result, we understand that people will be disappointed in PlatformIO Project because we can't implement all things in the nearest time and people don't like to wait...
[image]

# What do we have today...

* vendors work on the own boards/platforms
* vendors develop own frameworks, libs and create for them own build solutions
* vendors try to build own IDE for THE OWN products
* vendors require from you specific OS
* vendors require to install software which you don't like to use
* the list is long...

# Future

The 20 months were great experience for us. We are glad that we understand existing problems in embedded world and know how to improve it. But ...

* How about to improve this world together? 
* How about to invest into single ecosystem?

The success of PlatformIO will be the success for vendors and PlatformIO users:

* vendors will be focused on developing platforms https://github.com/platformio/platformio/issues/479 and save time/money
* users will have the single ecosystem (IDE, build software, library manager, hardware unit-testing, ...) for different hardware.

Just take a look at [PlatformIO 3.0](https://github.com/platformio/platformio/issues/423)...

# Invest in future

As we said, we love our community and want to discuss it with you. [PlatformIO team](https://github.com/orgs/platformio/people) is ready devote one's time to PlatformIO 3.0/IDE and make our favourite hobby(PlatformIO) as main work. It would be honour for us to make you happier working on the PlatformIO in the main time and have time for the rest, family and the new ideas.

In this situation, we propose the next versions for "Which should be PlatformIO 3.0 and PlatformIO IDE?":

1. FREE for all (including commercial using)
2. FREE for uncommercial and paid for commercial
3. Required DONATE (with minimal value) for uncommercial and paid for commercial

2 and 3 are very same. We don't know who are our users. 

In any case, we will work on PlatformIO and will implement all your requests. The only 1 difference: who will be investor of PlatformIO? We will be happy if you decide to invest in PlatformIO and we will be independent from external influence. You will decide which features should be implemented firstly, etc.

Sincerely, PlatformIO team:

* **Ivan Kravets** (@ivankravets), Ph.D, Researcher and Software Architect, Founder of @PlatformIO
*  **Valerii Koval** (@valeros), M.Sc. in Electronics  
*  **Dmytro Kyrychuk** (@orgkhnargh), Bachelor of Computer Science

g7uvw, 2016-02-01:
I'm a very happy user of PlatformIO and would happily try to get more involved in the development side of things, except I don't know where to start. Is there a hacking guide? "Edit here to add/change $foo" ?

I'm playing with the STM32 parts at the moment and I hope I may be able to contribute some example code for them.

ivankravets, 2016-02-01:
@g7uvw Thanks! Please take a look as #479. It will allow our users to develop own platforms and etc. I will document it.

supr, 2016-02-03:
Hi @ivankravets and team thanks for sharing PlatformIO with us. I very much loved using it. As i see it. I love actively working on the project in whatever aspects i could.

neuhalje, 2016-02-10:
Hi @ivankravets and team. I just started with embedded systems (arduino) development in my spare time. I am sure I have only seen the most superficial aspects of PlatformIO and I must say: It works and is useful. IMHO these are the most important aspects of any kind of tooling so consider this a compliment.

My 2cents on your question: pick option (1) and add the possibility to donate (but not mandatory). 

_Reasoning_: I think that giving free (-> speak) access to information is  a good thing to do, and you are in the position to do so.  Personally I would like to "give back" to the community, and I can do this by showing my appreciation via money - fine.

So, although I use PlattformIO very little, I would use the "donate" button.

mogorman, 2016-02-10:
i am a avid platformio user, I package it for my distro, and have tried to be a positive member of the community.  It would disappoint me if it stopped being free and open source software.  I would be force to stop using it and go back to my makefiles and horrible cross environments.  I would be happy to donate and think you should try to encourage the board manufacturers that you facilitate use of to donate as well.  Best of luck in whatever you choose

mogorman, 2016-02-10:
also I don't know how many people are like me, but i love the platformio tool, but have no real use for the ide.

boneskull, 2016-02-19:
@ivankravets + team,

As @mogorman wrote, I have no use for an IDE either. 

The PlatformIO "platform" is excellent and completely usable without such a thing.  So if it were up to me--which it isn't, because I'm not an investor--in order of priority:

1.  Sink resources into plugin development for JetBrains, Visual Studio, Sublime, Atom, etc.
2.  Work to keep the registry reliable and scalable.
3.  Continue excellent work on the CLI.

I don't know anything about investing or business, but my intuition suggests:

- Businesses will want to see a **reliable** service that integrates with their *existing* tools, and
- Investors will be more impressed by user interfaces than CLI's

Keep up the good work!

cbentle, 2016-03-08:
I love platformio. I am new to development in general, and I have easily downloaded 15 tool chains, ide's, etc for development of IOT. And after all of that I finally found PlatformIO and wish that I found it first. I have only been using it for a week now, and I have seen the donate button a few times now, and I click remember because; A. I need another paycheck, and B. I want to give it a good trial before I jump into support. I don't want to say thank you for everything your team has done, the instructions are difficult at first, but I only think that is because I am new to developing. But the instructions are far more easier to get than majority of other IDE. I am just now starting my own blog for beginners www.reddogalpha.com and I will share this tool as the best option for beginners, with arduino ide second. When I get my next paycheck I will donate,  Android when I am more competent I will develop PlatformIO IDE with the team. Thanks.

ivankravets, 2016-04-12:
Dear PlatformIO Users,

We have just answered the most questions in the second part of this discussion. Please read it and make a feedback about PlatformIO Club.

Thanks in advance!

[Continued in Part II…](https://community.platformio.org/t/urgent-further-fate-of-platformio-project-part-ii/148)

closed 2017-11-29 by ivankravets