platformio/platformio-core #491: Arduino framework does not work with network boot.

rwalli, 2016-02-01:
Code generated with platformio-2.8.2 or platformio-3.0.0dev fails to work with Ariadne or Netload bootloaders.
If i compile with various versions of the Arduino IDE the upload over ethernet works.
see also https://github.com/codebendercc/Ariadne-Bootloader/issues/22

Robert

rwalli, 2016-02-02:
Update: found out that the issue results from binary validation in the bootloader but i can't figure out why
PlattfromIO produces different kind of binaries with the same compiler.
Could you please delete this topic until i find more details.

closed 2016-02-02 by rwalli

ivankravets, 2016-02-02:
PlatformIO uses own code builder and build flow. We don't hide what we build. You can see all commands in the build output. Please contact with authors of the bootloader.