platformio/platformio-core #492: Coding including Audio.h with teensy31 board fails to compile from commandline

mattb, 2016-02-02:
On my Mac, the following steps result in the compilation failure below:

    $ platformio init -b teensy31
    $ echo '#include <Audio.h>' > src/main.cpp
    $ platformio run

The output is:

    [Mon Feb  1 19:47:51 2016] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
    ----------------------------------------------------------------------------------------------------------------------------------------
    arm-none-eabi-g++ -o .pioenvs/teensy31/src/main.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -DPLATFORMIO=020802 -I.pioenvs/teensy31/FrameworkArduino -I.pioenvs/teensy31/Audio src/main.cpp
    arm-none-eabi-g++ -o .pioenvs/teensy31/FrameworkArduino/AudioStream.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino .pioenvs/teensy31/FrameworkArduino/AudioStream.cpp
    arm-none-eabi-g++ -o .pioenvs/teensy31/FrameworkArduino/DMAChannel.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino .pioenvs/teensy31/FrameworkArduino/DMAChannel.cpp
    arm-none-eabi-g++ -o .pioenvs/teensy31/FrameworkArduino/HardwareSerial1.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -fsingle-precision-constant -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino .pioenvs/teensy31/FrameworkArduino/HardwareSerial1.cpp
    In file included from .pioenvs/teensy31/Audio/Audio.h:16:0,
    from src/main.cpp:1:
    .pioenvs/teensy31/Audio/DAC.h:21:16: error: expected ')' before '*' token
    DACClass(Dacc *_dac, uint32_t _dacId, IRQn_Type _isrId) :
    ^
    .pioenvs/teensy31/Audio/DAC.h:35:2: error: 'Dacc' does not name a type
    Dacc *dac;
    ^
    .pioenvs/teensy31/Audio/DAC.h:37:2: error: 'IRQn_Type' does not name a type
    IRQn_Type isrId;
    ^
    .pioenvs/teensy31/Audio/DAC.h: In member function 'void DACClass::enableInterrupts()':
    .pioenvs/teensy31/Audio/DAC.h:31:44: error: 'isrId' was not declared in this scope
    void enableInterrupts()  { NVIC_EnableIRQ(isrId); };
    ^
    .pioenvs/teensy31/Audio/DAC.h:31:49: error: 'NVIC_EnableIRQ' was not declared in this scope
    void enableInterrupts()  { NVIC_EnableIRQ(isrId); };
    ^
    .pioenvs/teensy31/Audio/DAC.h: In member function 'void DACClass::disableInterrupts()':
    .pioenvs/teensy31/Audio/DAC.h:32:45: error: 'isrId' was not declared in this scope
    void disableInterrupts() { NVIC_DisableIRQ(isrId); };
    ^
    .pioenvs/teensy31/Audio/DAC.h:32:50: error: 'NVIC_DisableIRQ' was not declared in this scope
    void disableInterrupts() { NVIC_DisableIRQ(isrId); };
    ^
    In file included from src/main.cpp:1:0:
    .pioenvs/teensy31/Audio/Audio.h: In member function 'virtual size_t AudioClass::write(uint8_t)':
    .pioenvs/teensy31/Audio/Audio.h:25:82: warning: no return statement in function returning non-void [-Wreturn-type]
    virtual size_t write(uint8_t c)                         { /* not implemented */ };
    ^
    scons: *** [.pioenvs/teensy31/src/main.o] Error 1
    ===================================================== [ ERROR ] Took 0.81 seconds =====================================================

ivankravets, 2016-02-02:
Do you use the latest PlatformIO? Please provide sketch to reproduce this issue.

ivankravets, 2016-02-02:
You have to implement `setup` and `loop` prototypes

mattb, 2016-02-02:
This is with the latest platformIO installed via pip today. Even with setup() and loop() routines the same error occurs - it appears to be missing type definitions such as IRQn_Type from teensy3/core_cm4.h (from the Teensy core). Attaching a tarball of my minimal project.
[sketch.tar.gz](https://github.com/platformio/platformio/files/113456/sketch.tar.gz)

mattb, 2016-02-02:
note: I added setup() {} and loop() {} to my cpp file. If I remove #include <Audio.h> then it compiles, so the error seems to be unconnected to the lack of those definitions.

ivankravets, 2016-02-02:
Thanks! The issue is linked with invalid Arduino library structure. I've just made PR https://github.com/PaulStoffregen/Audio/pull/163 . Please wait while @PaulStoffregen merge it.

PaulStoffregen, 2016-02-02:
> Please wait while PaulStoffregen merge it.

done

ivankravets, 2016-02-02:
@PaulStoffregen thanks a lot!

@mattb please reinstall library http://platformio.org/#!/lib/show/70/Audio

```
platformio lib uninstall 70
platformio lib install 70
```

closed 2016-02-02 by ivankravets

mattb, 2016-02-03:
Thanks! Seems to be working great now.