platformio/platformio-core #493: MSP430 Toolchain does not work on Windows

nmaas87, 2016-02-02:
I did try to compile the Blink project example with the most recent version of Platformio in Atom.io in Windows 8.1 64 bit but got the problem that the Toolchain does not work for MSP430:
Error was

User\.platformio\packages\ldscripts" -L.pioenvs\lpmsp430g2553 -Wl,--start-group -lm .pioenvs\lpmsp430g2553\libFrameworkEnergia.a -Wl,--end-group
c:/users/User/.platformio/packages/toolchain-timsp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld.exe: cannot open linker script file memory.x: No such file or directory
collect2: ld returned 1 exit status
scons: *** [.pioenvs\lpmsp430g2553\firmware.elf] Error 1
 [ ERROR ] Took 2.63 seconds 

Any idea how to solve that?

nmaas87, 2016-02-03:
Ok,

the error is due to a blank sign in the userpath of Windows.
So if you got an User Path c:/users/My User/.. you get the error i described above,
if you got an User Account like MyUser it does work.
Maybe you need to " " the paths to fix the problem for good?

ivankravets, 2016-02-08:
@nmaas87 we use Energia's toolchain. The issue is linked with it. Try temporary solution https://github.com/platformio/platformio/issues/143#issuecomment-88060906

nmaas87, 2016-02-14:
Temporary Solution does work - however, I hope it is going to be fixed for real - soon. I wanted to hold an talk on Platformio and present it on a Raspberry Pi meet up - would be lovely not to need to include this hack in front of the audience :).

ivankravets, 2016-02-14:
> I hope it is going to be fixed for real - soon.

I changed this issue to `enhancement`, I hope we will find time for it.

> I wanted to hold an talk on Platformio and present it on a Raspberry Pi meet up - would be lovely not to need to include this hack in front of the audience :).

The issue is linked only with Windows users who have space in `%USERPROFILE%`. 

Where will be that meet up? We would be THANKFUL if you say something about PlatformIO. Do you need something from our side?

nmaas87, 2016-02-14:
The PiAndMore will be on the 11. of June at the University of Trier, Germany.
We normally get about 200 visitors, hobbyists and students who are into Raspberry Pi, Arduino, ESP8266 and Co. I wanted to deliever an 30 Minute talk about PlatformIO as I am part of the staff of PiAndMore and I find your project most interesting and I would love to get more people to use it :).
So I would basically just introduce the people to PlatformIO in my talk, show them how easy it is to get it up and running - including Atom IO - and yep, that should be it.
The talk will be held in german and I will post it online after the talk on my Website www.nico-maas.de - if you want something especially shown or such, do not hesitate to just write me @ mail@nico-maas.de :).

ivankravets, 2016-02-14:
@nmaas87 thanks a lot! Please e-mail me when your presentation will be ready. I'll take a look at it. I hope that we will finish PlatformIO 3.0 till Jun, 11 2016 :smile:

ivankravets, 2016-04-25:
Please try this temporary solution http://docs.platformio.org/en/latest/faq.html#windows-unicodedecodeerror-ascii-codec-can-t-decode-byte

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-timsp430/issues/2

closed 2016-11-27 by ivankravets