platformio/platformio-core #494: Update Pro Micro variants .h file

malachib, 2016-02-03:
I looked around but couldn't find a way to submit this change myself, so asking here - the sparkfun_promicro/pins_arduino is nearly a year out of date can we get that updated?  (I believe the source is here [https://github.com/sparkfun/SF32u4_boards/tree/master/sparkfun/avr/variants/promicro](url)

On a related note; is there a way for me to do something like a pull request to save you the hassle?

valeros, 2016-02-03:
Hi @malachib! Please try updated package via `platformio update`.

malachib, 2016-02-03:
OK I will try it out, it looks promising!  How about the related question, where I'd be able to submit the change myself?

valeros, 2016-02-03:
@malachib, your local `framework-arduinoavr` is located here `~/.platformio/packages/framework-arduinoavr` or for Windows here `%HOMEPATH%/.platformio/packages/framework-arduinoavr` and for now, there is no way to update global packages by users.

malachib, 2016-02-08:
For the record, your update seems to work - it compiles - although I've toasted all my 32u4 boards at the moment so can't do a "full test" just yet =)

ivankravets, 2016-02-08:
@malachib please don't forget to give us a star :star: https://github.com/platformio/platformio/stargazers

closed 2016-02-08 by ivankravets

malachib, 2016-02-08:
Hmm.  I just tried uploading to a non-toasted board and now it's toasted too, and looks like it failed the same way:
[toasted1.pdf](https://github.com/platformio/platformio/files/122091/toasted1.pdf)

maybe I needed to do a clean first?

ivankravets, 2016-02-08:
@malachib try `platformio run -t program` instead `-t upload`