platformio/platformio-core #496: Serial Monitor not connecting

vidia, 2016-02-05:
I have been running code on an ESP8266 NodeMCU board successfully from PlatformIO, but when I try to attach the serial monitor I simply get output like 

`$> /Users/david/.atom/packages/platformio-ide/penv/bin/platformio -f -c atom serialports monitor -p /dev/cu.SLAB_USBtoUART -b 115200`

`tab 1 of window id 12421` 

When connecting with the same settings from the arduino IDE, I get a proper connection so I am confident that it doesn't have anything to do with my board settings. 

Am I supposed to attach to that window id somewhere? I expected to see the output right in the serial monitor.

QuickSander, 2016-02-07:
Search for a Terminal window in your OS desktop environment with serial monitor started as a process in it. It might have opened behind some other window?

ivankravets, 2016-02-08:
@QuickSander thanks for the answer!

closed 2016-02-08 by ivankravets