platformio/platformio-core #497: How to specify library dependencies in platformio.ini

probonopd, 2016-02-06:
In my project, I need to specify as dependencies:

 * Adafruit_MQTT_Client from https://github.com/adafruit/Adafruit_MQTT_Library, `trunk` branch
 * NeoPixelBus from https://github.com/Makuna/NeoPixelBus/tree/UartDriven, `UartDriven` branch

How would I specify those? The specific branches are important.
It would be nice if the [documentation](http://docs.platformio.org/en/latest/projectconf.html#lib-install) could be made a bit more verbose on how to do this.

gepd, 2016-02-08:
You have to include [lib_dir](http://docs.platformio.org/en/latest/projectconf.html#lib-dir)

your platformio.ini should looks like:
```
[env:uno]
platform = atmelavr
framework = arduino
board = uno

[platformio]
lib_dir=path

```
the `path` should end in a folder where are included both libraries

is that what you are looking for?

probonopd, 2016-02-08:
Not sure. I was looking for something that would grab the libraries automatically for me from GitHub using `git`. Use case: I put a sketch online and I want my users to just download it, press run, and then ideally platformio would be smart enough to download its dependencies before attempting to compile.

ivankravets, 2016-02-08:
The first library is added http://platformio.org/#!/lib/show/1092/Adafruit-MQTT

As for the rest, see https://github.com/platformio/platformio/issues/461

closed 2016-02-08 by ivankravets