platformio/platformio-core #498: Include library dependencies using their GitHub URLs

probonopd, 2016-02-06:
Wouldn't it be nice if we could
`#include https://github.com/Makuna/NeoPixelBus/blob/814a4f0250d408b6f0b7252537352ce0b93e4b3d/src/NeoPixelBus.h`?

Golang seems to work like this. The result would be painless dependency management.

loopj, 2016-02-28:
This would effectively require code parsing and rewriting, which seems a little funky in c/c++ land. Go already has this concept built in to the language.

Having github urls in the code would make for very unportable code, and would tightly bind code to platformio's build system.

Just my 2c.

gerritv, 2016-02-28:
You assume that everything is connected to the internet all the time, not something you should count on. Also any notion of repeatable builds goes out the door. No serious development organization would use this. There is no Change Management (those parts of your code can change at the other parties whim while you are building),non repeatable builds, no sane possibility of source code control.

Plus not to mention that you expect platformio to parse your entire source code looking for #include lines (possibly inside #ifdef's!) before calling the compiler.

probonopd, 2016-02-28:
> Also any notion of repeatable builds goes out the door.

Not if you include a specific SHA-1 hash

ivankravets, 2016-07-16:
The parse source code and replace with real includes will bring a lot of problems to the different parts of PlatformIO. The integration with IDEs will not works. Code completion and code linting services will raise a lot of errors.

In any case, I'll allow defining of "dependency" as "git" URL.

commit 758396c9 (2016-08-02): PlatformIO Library Manager 3.0 // Resolve #588 , Resolve #414, Resolve #498, Resolve #475, Resolve #410, Resolve #461, Resolve #361, Resolve #507, Resolve #554
closed 2016-08-08 by ivankravets