platformio/platformio-core #499: Parse Arduino library.properties

probonopd, 2016-02-07:
Instead of requiring its own `library.json`, PlatformIO should parse Arduino `library.properties`. Thanks @ladyada for the suggestion in https://github.com/adafruit/Adafruit_MQTT_Library/issues/30#issuecomment-180874470

ivankravets, 2016-02-26:
> Instead of requiring its own library.json

This is not "OWN". This is cross platform manifest without depending to any proprietary framework or platform. This manifest will be extended and improved significantly in PlatformIO 3.0, such as: dependencies with ``semiver`` support, tests configuration, additional settings to build system.

This manifest is developed for PlatformIO ecosystem but can be used with any software/build system.

> Thanks @ladyada for the suggestion in adafruit/Adafruit_MQTT_Library#30 (comment)

I don't know why @ladyada focuses on Arduino and develops new hardware around Arduino things. We propose vendors to focus on the hardware development and save valuable time which they spent on creating complicated manuals how to "reanimate" Arduino editor for their boards.

PlatformIO provides professional instruments for rapid development including own IDE http://platformio.org/#!/platformio-ide . 

Nevertheless, @probonopd please don't disturb @adafruit and make PR to https://github.com/platformio/platformio-libmirror/tree/master/configs/adafruit .

ladyada, 2016-02-27:
"I don't know why @ladyada focuses only on Arduino and develops new hardware around Arduino things."

great question! we have about ~100 C/C++ libraries for sensors, displays, etc. i maintain them - and its a lot of work to do so :) the Arduino core is portable, lightweight, and has been adopted by many vendors. being able to say "this works with Arduino UNO" is powerful because its a ~10 years old, low cost, easy-to-test standard. 
given the vast work it is to maintain the libraries we do (and we have tons of PR & issues still pending) i must choose my battles.
libmirror is great, please keep doing that! if i dont have to maintain and support your startup's code, it makes my life easier and that means i can write more libraries :+1:

thanks! :)

ivankravets, 2016-03-05:
@ladyada 

> if i dont have to maintain and support your startup's code

I didn't ask @adafruit to maintain or support our code. We do complicated work without any support and totally for free. I only proposed to keep [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) manifest file inside your libraries. What is more, you don't need to create it because we make ready PR.

The "maintain and support" for [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) means to change `version` field when new release is out. You do the same step for `library.properties`.

------

I see that you don't know about it. I propose `library.json` to Arduino Community in 2014 year. See https://groups.google.com/a/arduino.cc/forum/#!searchin/developers/platformio/developers/xCDjekcQW4k/EK7ZtQaaEhoJ . They didn't accept it. Though, I had the support from the famous contributors of Arduino, such as @PaulStoffregen. In the final result, they "invented" this `library.properties` 1 year ago specially for own aims. Where is the "profit" for the whole embedded community?

Finally, I've changed my opinion as for @adafruit. You talk on the different magazines, youtube, articles that you help electronics beginner and newbie to embedded world. That is great! The problem here that @PlatformIO doesn't buy @adafruit products and we are not your customers. As result, you "are not ready to support us". 

Conclusion. We will keep own mirror with the manifests for your libraries. What is more, we will continue support your clients who bought your boards and want to use professional instruments for embedded development instead Arduino IDE:

* http://platformio.org/#!/boards
* https://github.com/platformio/platformio/issues?utf8=✓&q=huzzah

-------

P.S: I propose to return to this discussion when we will finish work on PlatformIO 3.0 with the improved library manager for [semver](https://github.com/platformio/platformio/issues/410) support, custom build environment and etc. See https://github.com/platformio/platformio/milestones/3.0.0

ivankravets, 2016-09-12:
Resolved in PlatformIO 3.0 Library Registry 
- http://platformio.org/lib

closed 2016-09-12 by ivankravets