platformio/platformio-core #500: compiled code doesnt seem to run on redbearlabblenano

mogorman, 2016-02-07:
i have a version 1.0 redbearlab ble nano.  I created simple blink script and had it upload it.  It just bricks my device.  I can reflash it with arduino boot loader or sample mbed blinky lights demo and it works fine.  I can also program it via arduino ide and have no problems with it.  Anything I can do to try to debug let me know.

mogorman, 2016-02-08:
i tried taking the ld linker script from the arduino code and used it in build flags so that i could try uploading the code using their bootloader. this also failed.  I don't think platformio is generating a proper intel hex to be uploaded.

mogorman, 2016-02-08:
ive also tried this on my mac and gnu/linux machine, as well as enabling sync flag on mounting of the drive just in case.

valeros, 2016-02-08:
Hi @mogorman! Thanks for your report! Could I ask you to test some [firmwares](https://dl.dropboxusercontent.com/u/10025728/ble_nano.zip) with your board?

mogorman, 2016-02-08:
yes i can.

mogorman, 2016-02-08:
i tried copying all of them to the board. none of them blinked the led on pin P0_19.  I didn't know what to look for aside from that to tell if it worked?  they did all take much longer to upload, like the intel hex was legit though.

valeros, 2016-02-09:
Please try this [version](https://dl.dropboxusercontent.com/u/10025728/ble_nano_blink_mbed.zip). Does it work?

mogorman, 2016-02-09:
it looks like this one attempts to flash but fails, when i reboot the board, it still is running the arduino bootloader from redbear labs

valeros, 2016-02-09:
Let's put aside all about Arduino. Do you have removable mass storage disk named MBED? Firmware from mbed cloud compiler works fine?

mogorman, 2016-02-09:
im not using arduino to flash. aside from one time i tried that way.  sorry for confusion.  I have been copying to the MBED storage device on my pc.  I haven't tried the cloud compiler.  let me set up account and try that now.

mogorman, 2016-02-09:
can confirm that mbed web compiler works with the following code
```
#include "mbed.h"

DigitalOut myled(LED1);

int main()
{
    while(1) {
        myled = 1;
        wait(1);
        myled = 0;
        wait(1);
    }
}
```

valeros, 2016-02-09:
Thanks. Could you test please this [hex](https://dl.dropboxusercontent.com/u/10025728/blinky.zip)? It's compiled from exported project for GCC toolchain.

mogorman, 2016-02-09:
did not seem to run.  i tested by flashing code that does not use led. then flashing this code. then after seeing this code did not work i flashed with my blink code i made in web compiler and confirmed it ran.

mogorman, 2016-02-09:
using there local builder it is producing a 14k hexfile compared to the 300k hexfile the web produced with same code

valeros, 2016-02-09:
Yep, looks like it's not PlatformIO problem. I will look for a workaround and let you know if there is any information. Thanks!

mogorman, 2016-02-09:
actually i take that back. it does work.  i didnt have srecord installed in environment.  it needed to combine the two hexs together

mogorman, 2016-02-09:
I just tested the output of platformio generated code when combined with the hex like they do and it worked 
```
srec_cat mbed/TARGET_RBLAB_BLENANO/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0/s130_nrf51_1.0.0_softdevice.hex -intel .pioenvs/redbear/firmware.hex -intel -o test.hex -intel --line-length=44

mogorman, 2016-02-09:
actually i just confirmed its same softdevice shipped with platformio

commit 9d1128af (2016-08-02): Implement firmware merging with base firmware for Nordic nRF51 development platform // Resolve #500 , Resolve #533
closed 2016-08-02 by ivankravets