platformio/platformio-core #501: [Espressif] setting board_f_cpu to 160mhz isn't working anymore

FWeinb, 2016-02-07:
Since update to latest version the `board_f_cpu` flag seams to be used for defining the flash frequency and 160 isn't supported:

```
"/Users/FWeinb/.platformio/packages/tool-esptool/esptool" -eo "/Users/FWeinb/.platformio/packages/framework-arduinoespressif/bootloaders/eboot/eboot.elf" -bo .pioenvs/nodemcu/firmware.bin -bm dio -bf 160 -bz 4M -bs .text -bp 4096 -ec -eo .pioenvs/nodemcu/firmware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
error: invalid flash frequency value: 160
```

It looks like that [this](https://github.com/platformio/platformio/commit/d658363b5307da5ce3b2f99abe605103a360a605) is the commit that breaks this.

ivankravets, 2016-02-10:
Please re-test with http://docs.platformio.org/en/latest/installation.html#d-development-version

closed 2016-02-10 by ivankravets

FWeinb, 2016-02-10:
This is working great! Thanks.

ivankravets, 2016-02-10:
@FWeinb Thanks for the report! :+1:

sticilface, 2016-02-11:
Works for me!  thanks

commit 6baf6568 (2016-02-10): Add "board_f_flash" option to Project Configuration File // Resolve #501