platformio/platformio-core #503: Unable to link C++ standard library to espressif platform build

martianmartin, 2016-02-09:
I am unable to link against stdc++ library using the espressfi platform. I am able to do this in arduino IDE by adding -lstdc++ to platforms.txt. I have tried adding "stdc++" and "supc++" to the LIBS array in the espressif.py builder but this does not seem to have any effect. Any assistance appreciated.

Thank you

valeros, 2016-02-09:
Hi @martianmartin! Did you try [build_flags option](http://docs.platformio.org/en/latest/projectconf.html#building-options) in your `platformio.ini`?
```
platform = espressif
framework = arduino
board = esp01
build_flags = -lstdc++ -lsupc++
```

martianmartin, 2016-02-09:
Hi @valeros, yes I tried using the build flags. When I do that, in addition to the 'undefined reference' errors I also get 'multiple definition' errors and 'dangerous reallocation: call0: call target out of range" errors. 

The multiple definition errors make me think that part of the library is being left out, so re-including the entire thing may cause overlap?

Let me know what other information I can provide you to clarify the problem. I didn't want to paste the whole compiler error printout because it's very long but I can if that would help.

martianmartin, 2016-02-09:
As an experiment I removed the lines which perform map operations, which allows it to compile successfully. Then, adding back:
`build_flags = -lstdc++ lsupc++`
results in the error:

> .pioenvs\esp12e\libFrameworkArduino.a(abi.o): In function `operator delete(void*)':
abi.cpp:(.text._ZdlPv+0x4): multiple definition of `operator delete(void*)'
c:/users/marzipan/appdata/roaming/spb_16.6/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libstdc++.a(del_op.o):d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/../../../../libstdc++-v3/libsupc++/del_op.cc:45: first defined here
collect2.exe: error: ld returned 1 exit status
scons: *** [.pioenvs\esp12e\firmware.elf] Error 1

valeros, 2016-02-09:
Thanks, @martianmartin. I've reproduced this error, it's due to library linking order. Could you please try this version of PlatformIO with your project?
`pip install https://github.com/platformio/platformio/archive/feature/cpp-libs-issue-503.zip`

martianmartin, 2016-02-10:
Yes, that worked, thank you so much. I am sincerely impressed with your responsiveness. 

Do you mind explaining how the library linking order was wrong and how you fixed it?

Cheers

closed 2016-02-10 by ivankravets

martianmartin, 2016-02-15:
Hi, the link that you provided with the fix is no longer available and the associated milestone has not yet been merged. I recently cleaned up my environment without saving this zip, may you please provide another link in whilst the change has not been merged? Thank you

commit 1a4d1c7d (2016-02-10): Fix unable to link C++ standard library to Espressif platform build // Resolve #503