platformio/platformio-core #506: Compiling difference between Arduino IDE and Platformio

ivankravets, 2016-02-10:
Moved from https://github.com/platformio/platformio-atom-ide/issues/38
-------
First, let me say thank you for this incredible Atom plugin!!! It majorly improves my dev time compared with the Arduino IDE!
I successfully compiled and uploaded a few programs for Arduino Due, but I ran into one bug.
The following program compiles, uploads and works with the Arduino IDE:
```c
char myBuffer[5] = {' ', ' ', ' ', ' ', ' '};
void setup() {}

void loop() {
  SerialUSB.println(fillMyBuffer());
  delay(1000);
}

char* fillMyBuffer() {
  for (int i = 0; i < 5; i++) {
    myBuffer[i] = 'A';
  }
  return myBuffer;
}
```

Using Atom + Platformio, I get the following build error:
error: 'fillMyBuffer' was not declared in this scope

Here's what's on the build panel:
```
Executing: platformio run 
[02/10/16 12:25:42] Processing dueUSB (platform: atmelsam, board: dueUSB, framework: arduino)

[02/10/16 12:25:42] Processing dueUSB (platform: atmelsam, board: dueUSB, framework: arduino)

arm-none-eabi-g++ -o .pioenvs\dueUSB\src\tmp_ino_to.o -c -fno-rtti -fno-exceptions -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DF_CPU=84000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -Dprintf=iprintf -D__SAM3X8E__ -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -DUSB_VID=0x2341 -DUSB_PID=0x003e "-DUSB_PRODUCT=\"Arduino Due\"" -DARDUINO=10606 -DPLATFORMIO=020804 -I.pioenvs\dueUSB\FrameworkArduino -I.pioenvs\dueUSB\FrameworkCMSISInc -I.pioenvs\dueUSB\FrameworkLibSam -I.pioenvs\dueUSB\FrameworkLibSam\include -I.pioenvs\dueUSB\FrameworkDeviceInc -I.pioenvs\dueUSB\FrameworkDeviceInc\3x8e\include -I.pioenvs\dueUSB\FrameworkArduinoVariant src\tmp_ino_to.cpp
arm-none-eabi-g++ -o .pioenvs\dueUSB\src\tmp_ino_to.o -c -fno-rtti -fno-exceptions -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DF_CPU=84000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -Dprintf=iprintf -D__SAM3X8E__ -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -DUSB_VID=0x2341 -DUSB_PID=0x003e "-DUSB_PRODUCT=\"Arduino Due\"" -DARDUINO=10606 -DPLATFORMIO=020804 -I.pioenvs\dueUSB\FrameworkArduino -I.pioenvs\dueUSB\FrameworkCMSISInc -I.pioenvs\dueUSB\FrameworkLibSam -I.pioenvs\dueUSB\FrameworkLibSam\include -I.pioenvs\dueUSB\FrameworkDeviceInc -I.pioenvs\dueUSB\FrameworkDeviceInc\3x8e\include -I.pioenvs\dueUSB\FrameworkArduinoVariant src\tmp_ino_to.cpp
AtomTest.ino: In function 'void loop()':
AtomTest.ino:7:32: error: 'fillMyBuffer' was not declared in this scope
AtomTest.ino: In function 'void loop()':
AtomTest.ino:7:32: error: 'fillMyBuffer' was not declared in this scope
scons: *** [.pioenvs\dueUSB\src\tmp_ino_to.o] Error 1
 [ ERROR ] Took 2.92 seconds 
scons: *** [.pioenvs\dueUSB\src\tmp_ino_to.o] Error 1
 [ ERROR ] Took 2.88 seconds 
```

Thanks again!

closed 2016-02-10 by ivankravets

mbariola, 2016-12-23:
Hello, I seem to have the same problem on Windows 10 64-bit, just installed platformio IDE 1.2.1 core 3.2.1 with clang & python 2.7, all is updated. 

What can I provide / how to troubleshoot? Compilation works perfectly fine on Arduino IDE. 

Error: C:/Users/Massi/Dropbox/Arduino, RPi, Adafruit/test/m0_mudra_bleuart_v1.1/m0_mudra_bleuart_v1.1/m0_mudra_bleuart_v1.1.ino:168:51: error: 'isUSBOn' was not declared in this scope

Many thanks

mbariola, 2016-12-23:
Moving methods loop() and setup() at the very end of the .ino file, after all accessory methods, seems to have resolved the issue. Still, a discrepancy between Arduino IDE and Platformio. I suspect it's some build flag re. forward reference but cannot find where to look for it ...

ivankravets, 2016-12-24:
@mbariola Could you provide INO file to reproduce this issue? It looks that INO to CPP converter doesn't work correctly in your case. Thanks.

mbariola, 2016-12-24:
@ivankravets  sure, is there a way to send it privately to you? it is a project I want to develop commercially and so I'd like to not attach it publicly here

ivankravets, 2016-12-24:
Of course, please send to me@ikravets.com. I'll not share it.

mbariola, 2016-12-24:
Thanks @ivankravets , just sent.

ivankravets, 2016-12-26:
@mbariola Thanks a lot for provided demo project. Our INO to CPP is VERY STRICT. See 
* https://github.com/platformio/platformio-core/blob/develop/platformio/builder/tools/piomisc.py#L34:L40

Please remove spaces/tabs before `bool isUSBOn()`.

ivankravets, 2016-12-26:
@mbariola Please note that I deleted your email and all tracks related to your project.

mbariola, 2016-12-26:
@ivankravets so it was a spurious space character? wow, I'd have never guessed (nor found) that. thanks for your troubleshooting, confidentiality and quick answer. Now I know what to look for if it ever pops up again. cheers!

ivankravets, 2016-12-26:
@mbariola yes, we don't catch spaces before types of prototypes. It will break general regular expression.

A few recommendations to your project:

1. Please remove `lib_dir` from `[platformio]` section in `platformio.ini`
2. Use PlatformIO's Library Manager and `lib_deps` option
* http://docs.platformio.org/en/latest/librarymanager/index.html

mbariola, 2016-12-26:
ok, I can't say I comprehend the why of what you suggest me to do as I picked platformio just 3 days ago and had used arduino.cc's manager to not change too much in one go, but I'll clone the project to be sure and follow your instructions. it'll help me in familiarizing with the platform. thanks!

mbariola, 2016-12-26:
OK, done, took just a few minutes and properly separating general libs such as the adafruit bluetooth one from the project specific ones such as BluefruitConfig.h. Thanks!

ivankravets, 2016-12-26:
Happy coding with PlatformIO! 😊

ivankravets, 2016-12-26:
@mbariola would be thankful for a star 🌟 
* https://github.com/platformio/platformio-core/stargazers

mbariola, 2016-12-26:
gladly, how would I do it? star your profile .. this issue .. the project ...?

ivankravets, 2016-12-26:
> star ... the project 

Yes, please use "🌟  Star" button above.

mbariola, 2016-12-26:
Done! Cheers :-)

commit 851f7db8 (2016-02-10): Fix issue with pointer (char* myfunc()) while converting from *.ino to *.cpp // Resolve #506