platformio/platformio-core #507: add nrf51822 ble library.

mogorman, 2016-02-11:
the examples linked from the nrf51822 library do not compile as we do not have the ble_api library packaged.  I added it to my lib folder and then ran platformio run and got stuck here.

```
[Wed Feb 10 23:05:14 2016] Processing redBearLabBLENano (platform: nordicnrf51, lib_install: 378, board: redBearLabBLENano, framework: mbed)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/redBearLabBLENano/src/main.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DNRF51 -DTARGET_CORTEX_M -DTOOLCHAIN_GCC_ARM -DTARGET_NRF51822 -DARM_MATH_CM0 -D__MBED__=1 -DTARGET_RBLAB_BLENANO -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DTARGET_MCU_NRF51_16K_S130 -DMBED_BUILD_TIMESTAMP=1448299814.51 -DPLATFORMIO=020803 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc248832578 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-213564679 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-250686161 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-638156104 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-677489754 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc41765840 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc1263518378 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1726842523 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-704867372 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-670779565 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-481778965 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc814346659 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-919766688 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1735742204 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc250697636 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-673340142 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-831684529 -I.pioenvs/redBearLabBLENano/BLE_API src/main.cpp
arm-none-eabi-g++ -o .pioenvs/redBearLabBLENano/BLE_API/source/BLE.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DNRF51 -DTARGET_CORTEX_M -DTOOLCHAIN_GCC_ARM -DTARGET_NRF51822 -DARM_MATH_CM0 -D__MBED__=1 -DTARGET_RBLAB_BLENANO -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DTARGET_MCU_NRF51_16K_S130 -DMBED_BUILD_TIMESTAMP=1448299814.51 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc248832578 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-213564679 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-250686161 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-638156104 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-677489754 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc41765840 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc1263518378 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1726842523 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-704867372 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-670779565 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-481778965 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc814346659 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-919766688 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1735742204 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc250697636 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-673340142 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-831684529 -I.pioenvs/redBearLabBLENano/BLE_API .pioenvs/redBearLabBLENano/BLE_API/source/BLE.cpp
arm-none-eabi-g++ -o .pioenvs/redBearLabBLENano/BLE_API/source/services/DFUService.o -c -std=gnu++98 -fno-rtti -fdata-sections -ffunction-sections -Wno-unused-parameter -fno-exceptions -Wextra -fno-delete-null-pointer-checks -fmessage-length=0 -mthumb -Wno-missing-field-initializers -c -fno-builtin -O2 -fomit-frame-pointer -Wall -MMD -mcpu=cortex-m0 -D__CORTEX_M0 -DTARGET_MCU_NRF51_16K -DTOOLCHAIN_GCC -DTARGET_MCU_NRF51822 -DNRF51 -DTARGET_CORTEX_M -DTOOLCHAIN_GCC_ARM -DTARGET_NRF51822 -DARM_MATH_CM0 -D__MBED__=1 -DTARGET_RBLAB_BLENANO -DTARGET_NORDIC -DTARGET_MCU_NRF51 -DTARGET_MCU_NORDIC_16K -DTARGET_M0 -DTARGET_MCU_NRF51_16K_S130 -DMBED_BUILD_TIMESTAMP=1448299814.51 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc248832578 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-213564679 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-250686161 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-638156104 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-677489754 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc41765840 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc1263518378 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1726842523 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-704867372 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-670779565 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-481778965 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc814346659 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-919766688 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-1735742204 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc250697636 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-673340142 -I.pioenvs/redBearLabBLENano/FrameworkMbedInc-831684529 -I.pioenvs/redBearLabBLENano/BLE_API .pioenvs/redBearLabBLENano/BLE_API/source/services/DFUService.cpp
In file included from .pioenvs/redBearLabBLENano/BLE_API/source/services/DFUService.cpp:19:0:
.pioenvs/redBearLabBLENano/BLE_API/ble/services/DFUService.h:26:29: fatal error: dfu_app_handler.h: No such file or directory
#include "dfu_app_handler.h"
^
compilation terminated.
scons: *** [.pioenvs/redBearLabBLENano/BLE_API/source/services/DFUService.o] Error 1

```

mogorman, 2016-02-11:
i put that set of libraries in lib as well but it does not seem so solve the problem

mogorman, 2016-02-11:
http://github.com/mogorman/renfield is what im working with.  currently it doesnt seem so see dfu lib i put in

mogorman, 2016-02-11:
i also forced it to not depend on dfu by commenting out a few blocks where it tried to add that service.  the out put i got out after that was 
```
arm-none-eabi-ranlib .pioenvs/redBearLabBLENano/libBLE_API.a
arm-none-eabi-g++ -o .pioenvs/redBearLabBLENano/firmware.elf -Wl,--gc-sections -Wl,--wrap,main -mcpu=cortex-m0 -mthumb --specs=nano.specs -Wl,-T"/home/mog/.platformio/packages/framework-mbed/variant/RBLAB_BLENANO/mbed/TARGET_RBLAB_BLENANO/TOOLCHAIN_GCC_ARM/NRF51822.ld" .pioenvs/redBearLabBLENano/src/main.o -L/home/mog/.platformio/packages/ldscripts -L/home/mog/.platformio/packages/framework-mbed/variant/RBLAB_BLENANO/mbed/TARGET_RBLAB_BLENANO/TOOLCHAIN_GCC_ARM -L.pioenvs/redBearLabBLENano -Wl,--start-group -lstdc++ -lsupc++ -lnosys -lmbed -lc -lgcc .pioenvs/redBearLabBLENano/libFrameworkMbed.a -lc -lgcc -lm .pioenvs/redBearLabBLENano/libBLE_API.a -Wl,--end-group
.pioenvs/redBearLabBLENano/libBLE_API.a(BLE.o): In function `BLE::BLE(unsigned int)':
BLE.cpp:(.text._ZN3BLEC2Ej+0x20): undefined reference to `createBLEInstance()'
```

mogorman, 2016-02-11:
library isnt being built correctly it seems

matoushybl, 2016-08-11:
Hi, I am having similar problems with BLE_API, many errors about No such file or directory, I have tested with both versions - 2 and 3 of platformio but without any success, could you please help me with it?

ivankravets, 2016-08-11:
@matoushybl could you share somewhere demo project to reproduce this issue?

matoushybl, 2016-08-11:
Here it is. https://github.com/matoushybl/Bluetooth-reproducer
the errors are
```
.piolibdeps/BLE_API_ID296/source/BLE.cpp:17:21: fatal error: ble/BLE.h: No such file or directory
#include "ble/BLE.h"
^
compilation terminated.
.piolibdeps/BLE_API_ID296/source/BLE.cpp:17:21: fatal error: ble/BLE.h: No such file or directory
#include "ble/BLE.h"

^
compilation terminated.
.piolibdeps/BLE_API_ID296/source/DiscoveredCharacteristic.cpp:17:42: fatal error: ble/DiscoveredCharacteristic.h: No such file or 
directory
#include "ble/DiscoveredCharacteristic.h"
^
compilation terminated.
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/BLE.o] Error 1
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/DiscoveredCharacteristic.o] Error 1
.piolibdeps/BLE_API_ID296/source/DiscoveredCharacteristic.cpp:17:42: fatal error: ble/DiscoveredCharacteristic.h: No such file or 
directory
#include "ble/DiscoveredCharacteristic.h"
^
compilation terminated.
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/BLE.o] Error 1
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/DiscoveredCharacteristic.o] Error 1
.piolibdeps/BLE_API_ID296/source/GapScanningParams.cpp:17:21: fatal error: ble/Gap.h: No such file or directory
#include "ble/Gap.h"
^
compilation terminated.
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/GapScanningParams.o] Error 1
.piolibdeps/BLE_API_ID296/source/GapScanningParams.cpp:17:21: fatal error: ble/Gap.h: No such file or directory
#include "ble/Gap.h"
^
compilation terminated.
scons: *** [.pioenvs/nrf51_mkit/lib/BLE_API/source/GapScanningParams.o] Error 1
src/main.cpp:2:21: fatal error: ble/BLE.h: No such file or directory
```

matoushybl, 2016-08-12:
@ivankravets Hi I managed to fix the BLE import errors by moving the contents of source directory a level higher, but I am still not able to get it compile. From my point of view, it seems like there si something wrong with the paths, it looks like the nRF51822_ID378 is not imported at all, I am not able to access the `app_dfu_handler.h` as the OP stated and other headers and so on such as `pstorage.h`. Is there some problem on my end or is that a bug in the project?

ivankravets, 2016-08-12:
@matoushybl need to update `library.json` manifests for the libraries. I'll do it later and close this issue.

Nevertheless, you try it manually. See docs for [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html#build) and `build.flags` option. See example with [microbit-dal](http://platformio.org/lib/show/360/microbit-dal). Click on `library.json` tab. You should provide the same includes in `library.json` manifest.

ivankravets, 2016-08-26:
@matoushybl the libraries have been updated. I tried [your project](https://github.com/matoushybl/Bluetooth-reproducer) with this config
```ini
[env:nrf51_mkit]
platform = nordicnrf51
framework = mbed
board = nrf51_mkit
lib_deps = ble-nrf51822
```

Looks, that SDK is broken. We had the similar issue here https://github.com/platformio/platformio/issues/709.

@palfrey , could you help us? Thanks.

palfrey, 2016-08-26:
@ivankravets: It needs a version of the nrf51-sdk with https://github.com/palfrey/nrf51-sdk/commit/fab54eae10079738ba06032f9181cc77e14179c8 added. It can't use that repository directly however as that's branched from the Lancaster University fork of nrf51-sdk, which is needed for the Micro:bit work from #709.

ivankravets, 2016-08-26:
@palfrey please sorry! I've just checked our new builder for ARM mbed and this project compiles without problems. It looks that these ARM mbed official libraries require the latest ARM mbed SDK.

@valeros please merge new builder to our development branch.

ivankravets, 2016-08-26:
@palfrey see our announce https://community.platformio.org/t/platformio-3-0-improved-integration-with-arm-mbed-ecosystem/639/11

Could you try new build system with microbit project?
1. `pio platform uninstall nordicnrf51`
2. `pio platform install https://github.com/platformio/platform-nordicnrf51.git#feature/new-mbed-builder-integration`

Thanks!

palfrey, 2016-08-26:
@ivankravets: That works, thanks!

ivankravets, 2016-08-26:
Can't wait when we release 3.0 😊

ivankravets, 2016-09-01:
@matoushybl fixed in PlatformIO 3.0 http://docs.platformio.org/en/stable/installation.html#development-version

closed 2016-09-01 by ivankravets

vkolotov, 2017-01-08:
Hi All, I'm still getting this:

`[Mon Jan  9 09:37:06 2017] Processing redBearLabBLENano (platform: nordicnrf51, board: redBearLabBLENano, framework: mbed)
--------------------------------------------------------------------------------
PackageManager: Installing framework-mbed @ ~3.131.0
Downloading...
Unpacking...
framework-mbed @ 3.131.0 has been successfully installed!

Verbose mode can be enabled via `-v, --verbose` option
Collected 18 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <ble> v2.7.0
|   |-- <compiler-polyfill> v1.3.0
Compiling .pioenvs/redBearLabBLENano/lib/ble_ID296/services/DFUService.o
In file included from .piolibdeps/ble_ID296/source/services/DFUService.cpp:19:0:
.piolibdeps/ble_ID296/ble/services/DFUService.h:26:29: fatal error: dfu_app_handler.h: No such file or directory
#include "dfu_app_handler.h"
^
compilation terminated.
*** [.pioenvs/redBearLabBLENano/lib/ble_ID296/services/DFUService.o] Error 1
========================== [ERROR] Took 16.98 seconds ==========================
`

What do I do?

ivankravets, 2017-01-11:
@vkolotov please move your issue here https://github.com/platformio/platform-nordicnrf51/issues

commit 758396c9 (2016-08-02): PlatformIO Library Manager 3.0 // Resolve #588 , Resolve #414, Resolve #498, Resolve #475, Resolve #410, Resolve #461, Resolve #361, Resolve #507, Resolve #554