platformio/platformio-core #508: mbed Seeeduino Arch Pro: problem with printf of floats

sstaub, 2016-02-11:
I'm using a Seeeduino Arch Pro on mbed. I try to use printf of float values, but it doesn't work.
When I use the same code with the mbed online compiler it works. I know that there is no functionality 
of print methods with floats on AVR/Arduino, but not on mbed. 
Here the code:

`// mbed AnalogIn Example

#include "mbed.h"

Serial debug(USBTX, USBRX); // tx, rx
AnalogIn fader(A0);
DigitalOut led1(LED1);
DigitalOut led2(LED2);
DigitalOut led3(LED3);
DigitalOut led4(LED4);

int main() {
  while (1) {
    led1 = (fader > 0.2f) ? 1 : 0;
    led2 = (fader > 0.4f) ? 1 : 0;
    led3 = (fader > 0.6f) ? 1 : 0;
    led4 = (fader > 0.8f) ? 1 : 0;
    debug.printf("percentage: %3.2f% %\n", fader.read()*100.0f);
    debug.printf("normalized: 0x%04X \n", fader.read_u16());
    wait_ms(200);
    }
  }`

valeros, 2016-02-11:
Hi @sstaub, thanks for your report. Could you please try to add this option to your `platformio.ini`:
`build_flags = "-Wl,-u _printf_float -Wl,-u _scanf_float"`  
```
[env:seeeduinoArchPro]
platform = nxplpc
framework = mbed
board = seeeduinoArchPro
build_flags = "-Wl,-u _printf_float -Wl,-u _scanf_float"
```

sstaub, 2016-02-11:
Changed it, but it have no effect.

valeros, 2016-02-11:
Thanks, I updated my comment above, please retest.

sstaub, 2016-02-11:
No it works. The changes should cover all formatted I/O commands of stdio.h like sprintf......

ivankravets, 2016-02-15:
@sstaub Please reopen this issue if you still need any help.

closed 2016-02-15 by ivankravets

sstaub, 2016-02-15:
I think there should generally a solution that the all the formatted I/O commands with floats should work like in the mbed online-IDE, without setting flags in the .ini file.

ivankravets, 2016-02-16:
@sstaub we will update mbed framework when https://github.com/mbedmicro/mbed/issues/1549 will be fixed.