platformio/platformio-core #510: URGENT: Dear PlatformIO 2.0 contributors...

ivankravets, 2016-02-11:
Dear,
* @ZachMassia
* @mplewis
* @franciscoruiz
* @WillemMali
* @twpayne
* @floretan
* @TimJay
* @sticilface
* @dh1tw
* @bkudria
* @ctag
* @Lutino
* @jrobeson
* @dkuku

Thanks a lot for your [contributions](https://github.com/platformio/platformio/graphs/contributors). We are going to continue work on [PlatformIO 3.0](https://github.com/platformio/platformio/issues/423). According to https://github.com/platformio/platformio/issues/490 we decided to keep FREE & OPEN SOURCE PlatformIO 3.0. However, the license will be changed 

form **Apache 2.0** 

```
Copyright 2014-2016 Ivan Kravets <me@ikravets.com>

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

   http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
```
to **GPL 2.0**
```
 Copyright (C) 2014-2016 Ivan Kravets. All rights reserved.
 
 This source file is free software; you can redistribute it and/or
 modify it under the terms of the GNU General Public License version 2
 as published by the Free Software Foundation.
 
 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 GNU General Public License for more details.
 
 You should have received a copy of the GNU General Public License along
 with this program; if not, write to the Free Software Foundation, Inc.,
 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
```

---------------

In this situation, we would ask you to transfer all copyrights for your ``code/modification/fix`` to Ivan Kravets @ivankravets. Please post here your comment with you decision: AGREE or NOT AGREE. 

If you are not agree with the next changes your ``code/modification/fix`` will be removed from PlatformIO 3.0.

Sincerely, PlatformIO Team.

ctag, 2016-02-11:
AGREE.

I'm really excited to see this! Looking forward to an awesome 3.0!

WillemMali, 2016-02-11:
AGREE.

I don't really like the GPL but I like this project. Have at it.

mogorman, 2016-02-11:
@ivankravets, is there a reason you didn't do gplv2 or later (aka gplv2+)? it would make platformio more compatible with free software ecosystem and still be gpl2 like you want

jrobeson, 2016-02-11:
also the FSF doesn't find GPL 2.0 and ANY Apache 2 licensed code to used together. However GPL 3.0 is compatible with Apache 2.0. Certainly there's no problem with this relicensing, but it would affect potential code reuse.  In any case, you might actually wanna consider the AGPL instead, because some people might build a platform.io builder rest API, and under the GPL they wouldn't be required to share any of the non GPL modifications they make to it, but with the AGPL they would be.

ZachMassia, 2016-02-11:
AGREE.

Looking forward to help out again once I'm done with these classes I'm taking.

Lutino, 2016-02-11:
Agree. It's my pleasure to help this great project move forward.

TimJay, 2016-02-12:
I disagree with URGENT, a license decision should not be rushed. 
Why GPL and not something more ecosystem friendly? EPL for example has a very good balance between open source feedback and business friendlyness. 
GPL makes any integration with commercial background very cumbersome. 
I will not oppose GPL for the little bits I contributed, but an explanation would be nice before I agree. 

Cheers
Tim

twpayne, 2016-02-12:
AGREE

bkudria, 2016-02-12:
AGREE

sticilface, 2016-02-12:
Agree

mplewis, 2016-02-12:
Agree. Glad to help!

ivankravets, 2016-02-12:
Dear @TimJay, @mogorman and @jrobeson,

Sorry for the misunderstanding. I've just update this issue with 

> UPDATE: The license will be changed THE ONLY for PlatformIO Core. The all source code which has relation to build scripts, boards and frameworks will be moved to separated repositories under Apache 2.0 license. More details #479 

The main idea is to move the code which are interesting for contributors to the separate repos and keep FULL control on the PlatformIO Core. 

If you know the situation where PlatformIO 3.0 under GPL 2.0 will no be compatible with something (I don't know with what), please explain here.

Thanks.

TimJay, 2016-02-12:
This quote from the Wikipedia page on license compatibility sums it up quite well:
'Flask developer Armin Ronacher draw in July 2013 a negative resume on the GPL compatibility in the FOSS ecosystem when he concluded:"When the GPL is involved the complexities of licensing becomes a non fun version of a riddle."'
https://en.m.wikipedia.org/wiki/License_compatibility#GPL_compatibility

rue-mogorman, 2016-02-12:
I agree with @TimJay that using gpl makes it more likely that you just need to use gpl everywhere.  Which is why I am for changing to gplv2 or later for max compatibility that way. otherwise i think a simpler 2 clause bsd like apache or mit is what you want.  Personally I am for keeping things as free for the users as possible so gpl v2 or later works well

TimJay, 2016-02-12:
@rue-mogorman In my experience you have a better chance of establishing a lively ecosystem with a business friendly 2-clause/apache license as you suggested. EPL tends to be even better at that because it enforces code feedback while still being open to commercial endeavours.

ivankravets, 2016-02-12:
@TimJay PlatformIO isn't a framework or a SDK. I don't see problems for the end users. We are discussing here PlatformIO Core. What is more, I received 0 contributions to PlatformIO Core for the latest 20 months. It means, that no one is interested in developing PlatformIO for FREE. 

I'll write additional comment to #490 with our decision. PlatformIO is dear for me and I'll apply maximum efforts to keep it FREE and Open Source. However, each of should understand that we live in real life where each work should be paid. To have full influence to the PlatformIO project will allow PlatformIO Team to work on it for the full time.

mogorman, 2016-02-12:
I agree with @TimJay that using gpl makes it more likely that you just need to use gpl everywhere.  Which is why I am for changing to gplv2 or later for max compatibility that way. otherwise i think a simpler 2 clause bsd like apache or mit is what you want.  Personally I am for keeping things as free for the users as possible so gpl v2 or later works well

sorry made from work account submitting again

ivankravets, 2016-02-12:
Please explain me where THE END USER will have problems with PlatformIO under GPL 2.0.

jrobeson, 2016-02-12:
@ivankravets: there are no problems with that. I do think you should consider adopting the AGPL instead of the GPL though.

jrobeson, 2016-02-12:
@ivankravets : also GPL 2 and GPL 3 are not compatible with each other, so platformio extensions that allow GPL 3.0 code would not be allowed. Neither would Apache 2.0 based extensions

ivankravets, 2016-02-12:
Thanks guys,

Ok, I see that GPL 3.0 is good candidate for all of us. See [With which licenses is the GPLv3 compatible?](http://www.ifross.org/en/which-licenses-gplv3-compatible) In this case, we can keep PlatformIO Core under GPL 3.0 and you will have ability to write own development platforms, extensions without any problems using these [compatible licenses](http://www.ifross.org/en/which-licenses-gplv3-compatible).

@TimJay what do you think?

jrobeson, 2016-02-12:
@ivankravets : any reasons not to consider the AGPL 3?

ivankravets, 2016-02-12:
@jrobeson You are right. With AGPL v3 all of us will have access to all modified versions of PlatformIO http://choosealicense.com/licenses/. 

Thanks!

rue-mogorman, 2016-02-12:
@ivankravets awesome! and thanks for all hard work on project.  it has been very useful to me

ivankravets, 2016-02-12:
@rue-mogorman thanks that using PlatformIO :beer: 

Let's wait for @franciscoruiz, @floretan, @dh1tw, @dkuku

mogorman, 2016-02-12:
@ivankravets awesome! and thanks for all hard work on project. it has been very useful to me
[edit:ugh did it again]

ivankravets, 2016-02-12:
I've just updated comment with AGPL v3.0 https://github.com/platformio/platformio/issues/510#issue-133033781

WillemMali, 2016-02-12:
> @rue-mogorman <https://github.com/rue-mogorman> thanks that using
> PlatformIO [image: :beer:]
>
> Let's wait for @franciscoruiz <https://github.com/franciscoruiz>,
> @floretan <https://github.com/floretan>, @dh1tw <https://github.com/dh1tw>,
> @dkuku <https://github.com/dkuku>

I think we should repeat the agreement for good measure: all who agreed so
far only agreed to GPL2. You might say people "mostly agreed" or "agreed in
spirit" but that won't help you if you get sued or get a DMCA notice.

So: I agree (with a move to AGPL v3). As an aside, I'm pretty sure my vote
shouldn't matter as I contributed nothing to core. I wouldn't mind if you
took me off the contributers file.

ivankravets, 2016-02-12:
> I think we should repeat the agreement for good measure: all who agreed so
far only agreed to GPL2. You might say people "mostly agreed" or "agreed in
spirit" but that won't help you if you get sued or get a DMCA notice.

You are right! Friends, @ZachMassia @mplewis @franciscoruiz @WillemMali @twpayne @floretan @TimJay @sticilface @dh1tw @bkudria @ctag @Lutino @jrobeson @dkuku , please repeat the agreement as for AGPL v3.0. Looks like it is the best candidate for all of us.

> As an aside, I'm pretty sure my vote shouldn't matter as I contributed nothing to core. I wouldn't mind if you took me off the contributors file.

You are the contributor of PlatformIO 2.0 and it doesn't matter how many lines you are contributed and to which part of PlatformIO. I appreciate each PlatformIO's contributor. 

Nevertheless, I've not written this post to say that "IF YOU ARE NOT AGREE, I WILL DO WHAT I WANT". I like our community and it's valuable for me. As you can see, we have started from GPL 2.0 and finished on AGPL 3.0 :blush:

mplewis, 2016-02-12:
I agree regarding AGPL v3.0.

jrobeson, 2016-02-12:
i agree regarding AGPL v3.0.

bkudria, 2016-02-13:
I agree WRT AGPL as well.

Lutino, 2016-02-13:
I agree with AGPL v3.0 license

>
> I agree WRT AGPL as well.
>
>
>
> > i agree regarding AGPL v3.0.
> >
> > <
https://github.com/platformio/platformio/issues/510#issuecomment-183527082>

ctag, 2016-02-13:
I agree with AGPL v3.0 as well.

floretan, 2016-02-13:
I agree to with changing to any of the license changes mentioned in this thread so far.

TimJay, 2016-02-13:
I agree regarding the use of the AGPL v3.0 license.

franciscoruiz, 2016-02-13:
I agree.

Francisco Ruiz

ZachMassia, 2016-02-13:
AGREE

ivankravets, 2016-02-18:
Dear @twpayne, @sticilface, @dh1tw and @dkuku,

Are you agree with https://github.com/platformio/platformio/issues/510#issue-133033781 and **UPDATE 2** (The license above is updated to AGPL v3.0) ?

Thanks!

dkuku, 2016-02-18:
agree

dh1tw, 2016-02-18:
Agree

ivankravets, 2016-02-25:
Dear Contributors,

You will be surprised... but we have decided to keep current PlatformIO 2.0 license in PlatformIO 3.0. We will be maximum permissive to our PlatformIO ecosystem and our Partners.

# MORE OPEN SOURCE TO PLATFORMIO

PlatformIO 3.0 (as 2.0) will be released under `Apache v2.0` under the same `Copyright 2014-2016 Ivan Kravets <me@ikravets.com>`.

Please sorry that disturb you.

Regards, [PlatformIO Organization](http://platformio.org/#!/contact).

----------

We want to thank @TimJay and @Eclipse Foundation for the great discussion and professional consultation.

mplewis, 2016-02-26:
Thanks for keeping us updated.

loopj, 2016-02-28:
<3 the Apache + MIT licenses, thanks @ivankravets!

closed 2016-03-08 by ivankravets