platformio/platformio-core #511: How can I list available libraries without being horribly, horribly annoying.

fake-name, 2016-02-11:
I'm looking for a particular library for arduino, so I'm trying to grep through the available libraries.

However, for some insane reason, `platformio lib search -f arduino` seems dead-set on forcing an extremely annoying "Show next libraries? [y/N]:" prompt every 10 items. 

There are 235 arduino libraries.

If I set `platformio settings set enable_prompts no`, it *just* returns the first 10 items, and then exits.

1. Why on earth are you doing your own pagination. `less` and `more` have existed for that *before I was even born*.
2. Why doesn't `platformio settings set enable_prompts no` cause it to emit all the available libraries. Having it default to exit after just the first 10 is basically useless.

fake-name, 2016-02-11:
Piping `yes` into `platformio lib search -f arduino` is a hacky workaround. However, it seems to usually just result in an eventual 500 error.

Is the lib search command doing web requests? Wat? Why isn't there a locally cached package list? That's how basically every package manager ever works.

tcurdt, 2017-05-04:
Please please fix this.

ivankravets, 2017-05-04:
Guys, I don't know a good solution for this issue. Currently, PIO Library Registry contains ~1700 libraries. Each library has a lot of related information in DB. We use full-text search over these data. As you can see, the query results have a good relevance.

Of course, we can "download" whole information to each user's machine and search over these data using simple string matching or fuzzy search. I don't know what is better

A temporary solution is to use `--noninteractive` option for `pio lib search` command if you don't like to press `Y`.

Please sorry.

tcurdt, 2017-05-04:
IMO there two ways to tackle this:

1) Download a db to the local machine - like many of the package manager do.

2) If you want to keep the DB on the server that's of course possible/fine, too. Do the search on server and return the results (in one request).

Where the results come from is really an implementation detail and should not be related to the question of paging/interactive.

The `platformio lib search -f arduino --noninteractive` isn't really a workaround either. Frankly speaking I don't even see how it is useful at all. Automatically going through the list every x seconds? This is really awkward :)

fake-name, 2017-05-05:
Does `--noninteractive` return **ALL** the possible results? If it just truncates, that's not a solution for anything.

tcurdt, 2017-05-05:
@fake-name imagine having a robot that presses "Y" for you every few seconds.

fake-name, 2017-05-05:
@tcurdt - See my comment from a year ago:

> Piping `yes `into `platformio lib search -f arduino`

That's literally what I was doing.

tcurdt, 2017-05-05:
@fake-name then you now know a command line parameter for it ;)

but yes - there seems to be no proper way of interacting with a long list of search results in PIO core at the moment.

ivankravets, 2017-05-05:
Could someone explain me why do you need 2000 libraries in the console? Just curious.

PIO Core caches library queries, you can use it in offline mode.

tcurdt, 2017-05-05:
@ivankravets one just expects a certain behaviour from a CLI in a unix environment

```
platformio lib search -f arduino|less
platformio lib search -f arduino|grep
platformio lib search -f arduino > arduino.txt
```

...and frankly speaking I can just not come up with user experience/interaction story where the current `--noninteractive` behavior could be useful.

fake-name, 2017-05-06:
In my case, I remembered that there *was* an arduino library for what I was doing, but didn't remember the name, and the search didn't work for things like the chip name, etc... I wanted to be able to skim the available arduino libraries.