platformio/platformio-core #512: eclipse integration

deladriere, 2016-02-12:
I followed  [these instructions](http://docs.platformio.org/en/latest/ide/eclipse.html#integration) but Eclipse doesn't recognise  ino files as C++ file 

Menu:Projet>build Projet works and compile fine
but I don't have PlatformIO command like upload under Menu:Projet>build Projet

ivankravets, 2016-02-12:
1. Please take a look at [PlatformIO IDE](http://docs.platformio.org/en/latest/ide/atom.html)
2. Try "Right Click" on the project in Eclipse, then `Make Targets > Build...`

Also, you can add "Targets View" via `Eclipse Menu: Windows > Show View > Make Targets`

deladriere, 2016-02-13:
OK Platfformio commands works but Eclipse show errors for all Arduino specifics like pinMode, digitalWrite, OUTPUT, etc. 
What is missing ?

ivankravets, 2016-02-13:
Please add the line on the top of file
```c
#include <Arduino.h>
```

deladriere, 2016-02-13:
Thx it works

closed 2016-02-13 by ivankravets

dinvlad, 2016-03-21:
I'm seeing the same error even after inclusions, and multiple index rebuilds/Eclipse restarts with the Blink example. Other projects suffer the same fate. I'm using Eclipse Mars.2 - is it considered compatible with Platform.IO?

ivankravets, 2016-03-21:
@dinvlad What is the extension of your main file? It should be `*.cpp`.

dinvlad, 2016-03-21:
Yes, it's *.cpp. I also tried .ino but that didn't help. Thanks

ivankravets, 2016-03-23:
@dinvlad 

1. You should use `*.cpp`. Rename to `.cpp`
2. Try to rebuild C/C++ index . Menu: Project > C/C++ Index > Rebuild
3. Please give me your screenshot like mine
[image]

dinvlad, 2016-03-23:
Yes, I did rebuild indices many times. Sorry but in the end I've decided to use Visual Studio/Atmel Studio since they provide a smoother experience out of the box and more importantly, support debugging (at the cost of being Windows-specific). Also of note that Atom IDE didn't have this issue.

deladriere, 2016-03-23:
Also moving to Visual Micro to have the debugging working

ivankravets, 2016-03-23:
@deladriere which boards do you debug?

ivankravets, 2016-03-23:
@dinvlad we don't force to use PlatformIO or any IDE. However, we would be thankful if you help us to resolve your issue. Could you provide screenshot from step 3?

Thanks!

deladriere, 2016-03-23:
@ivankravets Arduino Zero

dinvlad, 2016-03-23:
@ivankravets Here it is. I'm using official Python distribution 2.7.10 64-bit, Eclipse Mars.2 (4.5.2), and PlatformIO installed using pip. I also installed CDT 8.6.0, though it wasn't required for it to work.

Btw it was a bit premature to say we completely moved to VS, since I'm still planning to use the excellent PlatformIO for CI with Jenkins, just not for IDE. Thanks a lot for your efforts!

[image]

ivankravets, 2016-03-24:
@deladriere we have OpenOCD package and you can debug your project in any IDEs which is listed in supported list by PlatformIO. If you are interested in it, we can help and explain. In any case, we are working with our IDEs Partners and PlatformIO users will receive debug feature very soon.

ivankravets, 2016-03-24:
@dinvlad Do you use the latest CDT plugin? I don't see [CDT Cross GCC Built-in Compiler Settings](https://cloud.githubusercontent.com/assets/1760209/13986353/a0b68be4-f10a-11e5-9555-30f0e14d911f.png).

Please download the latest [Eclipse IDE for C/C++ Developers](http://www.eclipse.org/downloads/packages/eclipse-ide-cc-developers/mars2) and try to generate new project.

dinvlad, 2016-03-24:
@ivankravets great! would it support "programmer-based" debugging, or purely software/USB-serial, and in the latter case would there be any limitations compared to the former (like the number of breakpoints, ability to trace etc.)?

There was no need to install a new copy of IDE (I had one configured for Java), I just enabled "CDT Optional Features->C/C++ GCC Cross Compiler Support" through "Install New Software" and had to check the cross-compilation flag from above manually (even after creating a fresh Eclipse project using pio from the command line), then copied the command string for cross-compiler from your screenshot and clicked "C/C++ Index->Freshen all files". The errors are now gone!

ivankravets, 2016-03-24:
@dinvlad 
>  would it support "programmer-based" debugging

Hardware-based debugging. GDB<->PlatformIO Debug Server via (OpenOCD, STLink, etc)

>  would there be any limitations compared to the former (like the number of breakpoints, ability to trace etc.)?

No any limitations. We are 100% open source.

> The errors are now gone!

Great! :+1: Thanks a lot that reported it. I hope that this information will be useful for our community.

Happy Coding! :blush:

dinvlad, 2016-03-24:
Sweet! It turns out that VisualMicro's Intellisense doesn't fully support Arduino 1.6.7 (yet) so I might be to square one with PlatformIO/Eclipse! Especially that it's getting the free hardware debugging support! Is that available on the development branch that I could test now?

ivankravets, 2016-03-26:
>  Is that available on the development branch that I could test now?

Which boards are you going to debug?

dinvlad, 2016-03-26:
@ivankravets Mega 2560 R3. I also have Atmel-ICE debugger if that helps (but could use another one if needed)

ivankravets, 2016-03-27:
@dinvlad 
Please read my article [Building and debugging Atmel AVR (Arduino-based) project using Eclipse IDE+PlatformIO](http://www.ikravets.com/computer-life/programming/2014/06/20/building-and-debugging-atmel-avr-arduino-based-project-using-eclipse-ideplatformio). Scroll to the bottom of article and use `avarice` instead `simulavr`. Also, take a look at [this article](http://avr-eclipse.sourceforge.net/wiki/index.php/Debugging).

GDB is located in `~/.platformio/packages/toolchain-atmelavr/bin/avr-gdb`.

dinvlad, 2016-03-29:
**ivankravets** thanks, very informative!

Unfortunately, it looks like my programmer is "too fresh" and is not yet supported by AVaRICE. Apparently it's using CMSIS-DAP/EDBG protocol to wrap USB communication, and that is thus far only supported by Atmel Studio and AVRdude. I got it working with both, but will be stuck with Atmel Studio for debugging for the time being. In any case, Eclipse/PlatformIO is my primary IDE now for the bulk of development!

dinvlad, 2016-03-29:
Hmm, does using CMSIS/EDBG mean that it's a unified interface that should be compatible with OpenOCD? Then this debugger could benefit from that new ecosystem instead of AVaRICE?

dinvlad, 2016-03-30:
Happy to report (partial) success!

Turns out AVaRICE has just introduced support for EDBG protocol on their development branch: http://svn.code.sf.net/p/avarice/code/trunk/avarice/NEWS

I've now successfully compiled it for Windows using bare Cygwin installation with its stock libusb-devel and libhidapi-devel packages and INSTALL-FROM-CVS readme from svn folder (however, kudos goes to http://www.avrfreaks.net/forum/atmel-ice-avarice-eclipse-and-debugging-ubuntu).

After that AVaRICE appears to start successfully (either from Cygwin shell or Windows CMD if you add C:\cygwin\bin to your PATH). For connection to EDBG, use the new '--edbg' key, e.g.
`avarice.exe --edbg --ignore-intr :4242`

Atmel-ICE and programmed device should then be detected automatically. There's no need to fuss around with libusb filters as suggested at the end of the installation, it just works! It's also helpful to keep Atmel Studio open in case an upload or debugging session 'bricks' the device (the installation steps from above don't break its compatibility).

The rest of instructions are the same as provided in your article.

Now I can start the debugging session from Eclipse and it even stops at an initial breakpoint, however I see messages about XML memory maps that I still have to troubleshoot. Hopefully that won't be much extra work though.

ivankravets, 2016-03-30:
@dinvlad great news! :+1: 

Please open separate issue to update AVaRICE binary in package. Also, please link to this issue.