platformio/platformio-core #513: unable install platformio on macos OS X 10.11.2 (15C50)

glebv, 2016-02-12:
unable install platformio on macos OS X 10.11.2 due to incorrect handling of installation of new pyserial 3.xx

`sudo pip install -U platformio`
Collecting platformio
  Downloading platformio-2.8.3-py27-none-any.whl (161kB)
... 
Installing collected packages: click, requests, pyserial, colorama, bottle, lockfile, platformio
  Found existing installation: pyserial 2.7
    Uninstalling pyserial-2.7:
      Successfully uninstalled pyserial-2.7
Successfully installed bottle click colorama lockfile platformio pyserial requests
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 9, in <module>
    load_entry_point('pip==8.0.2', 'console_scripts', 'pip')()
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/__init__.py", line 217, in main
    return command.main(cmd_args)
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/basecommand.py", line 246, in main
    pip_version_check(session)
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/utils/outdated.py", line 102, in pip_version_check
    installed_version = get_installed_version("pip")
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/utils/__init__.py", line 817, in get_installed_version
    working_set = pkg_resources.WorkingSet()
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 626, in __init__
    self.add_entry(entry)
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 682, in add_entry
    for dist in find_distributions(entry, True):
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 2080, in find_eggs_in_zip
    if metadata.has_metadata('PKG-INFO'):
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1610, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1968, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1848, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "/Library/Python/2.7/site-packages/pip-8.0.2-py2.7.egg/pip/_vendor/pkg_resources/__init__.py", line 1788, in load
    mtime = os.stat(path).st_mtime
OSError: [Errno 2] No such file or directory: '/Library/Python/2.7/site-packages/pyserial-2.7-py2.7.egg'
`~ $ python --version`
Python 2.7.10

ivankravets, 2016-02-12:
Please give me an output of 
```
pip list
```

glebv, 2016-02-12:
It's weird but it's true, 
`sudo pip uninstall platformio`
`sudo pip install -U platformio`
It solved the issue

ivankravets, 2016-02-12:
Hm :( I don't have comments :(

P.S: Have you tried [PlatformIO IDE](http://docs.platformio.org/en/latest/ide/atom.html)? It contains built-in PlatformIO CLI which could be installed globally to your system http://docs.platformio.org/en/latest/ide/atom.html#install-shell-commands

glebv, 2016-02-12:
Yes, I've installed it before platformio CLI, but it didn't install platformio as a global lib (at least it wasn't able use without full path) so I decided to install by pip.

ivankravets, 2016-02-12:
@glebv I recommend to remove global PlatformIO `sudo pip uninstall platformio` and make these steps https://github.com/platformio/platformio-atom-ide/issues/35#issuecomment-183431027

Thanks!

closed 2016-02-12 by ivankravets