platformio/platformio-core #516: program target doesn't supply port to avrdude

rnovacek, 2016-02-12:
I'm using platformio to program ATMega328p chip using Arduino Nano as programmer.

The command I'm using is following:
`platformio run -t program`

It gets translated to this:
`"/home/radek/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/home/radek/.platformio/packages/tool-avrdude/avrdude.conf" -c stk500v1 -U flash:w:.pioenvs/ATMega328p_8MHz/firmware.hex:i`

There is no serial port supplied to avrdude and the programming process fails because avrdude tries to use /dev/ttyS0 by default. I was not able to supply the port nor on commandline using `--upload-port` nor in the environment definition.

I've figured out that following patch fixes the issue for me (but I'm not sure if it won't break something else):
```patch
diff -up ./builder/scripts/atmelavr.py.fix ./builder/scripts/atmelavr.py
--- ./builder/scripts/atmelavr.py.fix   2016-02-12 22:11:00.452578300 +0100
+++ ./builder/scripts/atmelavr.py       2016-02-12 22:11:16.791028191 +0100
@@ -27,8 +27,8 @@ from platformio.util import get_serialpo
 
 def BeforeUpload(target, source, env):  # pylint: disable=W0613,W0621
 
-    if "program" in COMMAND_LINE_TARGETS:
-        return
+    #if "program" in COMMAND_LINE_TARGETS:
+    #    return
 
     if "micronucleus" in env['UPLOADER']:
         print "Please unplug/plug device ..."
```

ivankravets, 2016-02-12:
Please read docs and "Arduino as ISP" http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

`program` target does't append options/flags automatically. This target is intended for the case where you need specific flags/options.

closed 2016-02-12 by ivankravets

rnovacek, 2016-02-12:
Ah, I was not aware of that, thank you.