platformio/platformio-core #520: Please add support for some SAMD21 boards:

malachib, 2016-02-16:
* Sparkfun SAMD dev / mini
* Adafruit M0 feather series (I expect all of them would work as one single feather m0 target)

I've been using the zeroUSB profile to (very effectively!) program my SAMD with platformio but technically it's the wrong PID and might eventually cause issues.  Have not tried my adafruit yet

thirstyfish, 2016-02-22:
+1

Have you managed to get the Adafruit Feather m0 Wifi board working?
( I'm referring to this one: https://learn.adafruit.com/adafruit-feather-m0-wifi-atwinc1500?view=all )
I tried to get it to work with the zeroUSB config, but it was missing analog PINs I'm using, so I tried to hack it together by copying the relevant pieces over from
_~/Library/Arduino15/packages/adafruit/hardware/samd_ (e.g. the variant files), but then I had some other issues unfortunately trying to compile the Adafruit-GFX-Library:

`.pioenvs/zeroUSB/Adafruit-GFX-Library/fontconvert/fontconvert.c:23:22: fatal error: ft2build.h: No such file or directory`

malachib, 2016-02-22:
Haven't tried the Adafruit yet (mine is the coveted BLE version) but I expect similar results as yours

aethaniel, 2016-03-23:
@ivankravets, this ticket should not be labelled vendor:atmel :wink: as the issue seems to be in OSX, maybe around user rights to read/write on a system peripheral (cu.usbmodem1411 in this case).

The boards listed here are compatible with Arduino Zero, thus the label vendor:arduino or arch:sam should be added instead

malachib, 2016-05-12:
I believe I've discovered what aggravates this problem.  I observe similar behaviors on my debian machine.

I theorize that some component of the build chain attempts to invoke bossac with something like "/dev/cu.usbmodem1411" and fails because it must be formatted only as "cu.usbmodem1411".

ivankravets, 2016-05-12:
@malachib what do you see for `pio serialports list`?

malachib, 2016-05-12:
For the record, pio run -t upload frequently works on my debian machine (haven't been using it on the mac lately).  Once in a while it fails and seems to leave it in a state which I must run manual bossac to recover.  Debian results of `pio serialport list`

While the device is out of programming mode, we see:

```
/dev/ttyS0
----------
Hardware ID: PNP0501
Description: ttyS0

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=2341:804D LOCATION=2-2.1
Description: Arduino Zero

```

While in programming mode (put there on purpose, not accidentally reached), we see:

```
/dev/ttyS0
----------
Hardware ID: PNP0501
Description: ttyS0

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=1B4F:0D21 LOCATION=2-2.1
Description: ttyACM0

```

ivankravets, 2016-05-12:
PlatformIO uses `pio serialport list` results before upload. I can't imagine how you receive `cu.usbmodem1411` instead `/dev/cu.usbmodem1411`... 😢

malachib, 2016-05-12:
I'm thinking the opposite is happening.  BOSSAC expects no /dev/ prefix and maybe one is present sometimes?   Just a theory only

ivankravets, 2016-05-12:
> BOSSAC expects no /dev/ prefix

@valeros is it true?

malachib, 2016-05-20:
Another tidbit - on debian today.  I'm discovering sparkfun board seems to need up to a 20s ish delay after entering program mode before it will accept programming.  Also, today pio isn't able to program it at all - the "state" described above (7d ago) seems to be program mode - which pio always succeeds at getting the device into.

tabchas, 2016-06-01:
Also having problems uploading to Adafruit Feather. Any ideas of a fix? This is only reason I'm still using the regular Arduino IDE.

ivankravets, 2016-06-01:
@tabchas , @malachib , looks like the issue with uploading has been fixed today. Could you try http://docs.platformio.org/en/latest/installation.html#development-version ?

malachib, 2016-06-02:
Willdo - thanks Ivan!

tabchas, 2016-06-03:
@malachib Does it work for you?

I still get an error during upload.

malachib, 2016-06-03:
I am trying on my Mac right now and it's behaving well.  However:

* I'm not sure if I'm running the dev version of platformio here (@ivankravets how do I check?)
* Haven't tested on the Mac in a while.  So maybe a false positive.  Gonna try my linux box shortly

Side note: almost blew up my SAMD today.  It still works, so the hunt for where that smoke was coming from continues =)

ivankravets, 2016-06-03:
@tabchas what is your OS? I've pushed new updates to development version (2.9.3.dev0). Could you try new build? See instructions http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2016-06-03:
@malachib 
> I'm not sure if I'm running the dev version of platformio here

Do you use PlatformIO IDE? Check CLI version via `Menu: PlatformIO > Home screen`.

malachib, 2016-06-03:
OK:

Mac: platformio 2.9.2
Debian: platformio 2.9.3.dev0

Unfortunately Debian still not playing nice.  Mac passes basic tests (cool!).  Following is for Debian only:

# Scenario A

If I do `pio run -t upload` it will:

a) promptly put device into program mode 
b) error out with:

```
No device found on ttyACM0
scons: *** [upload] Error 1
```

at which point `pio run -t upload` will only generate 

```
Set binary mode
No device found on ttyACM0
Send auto-baud
Set binary mode
scons: *** [upload] Error 1
```

 so I must proceed to c) run manual bossac for successful upload (no sudo needed)

----
# Scenario B

If I do `sudo pio run -t upload` it sometimes manages to upload successfully, after trying a few times getting errors like:

```
Set binary mode
Send auto-baud
Set binary mode
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
version()=v1.1 [SparkFun:XYZ] Sep 24 2015 16:50:58
chipId=0x10010005
Connected at 115200 baud
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)

SAM-BA operation failed
```

Sometimes not, in which case I have to run my manual bossac many many times to coax it back into working again

tabchas, 2016-06-04:
Upload works! I selected "Arduino Zero - Native USB Port" and am using the Adafruit Feather Bluefruit M0 board.

Thanks guys!

ivankravets, 2016-06-04:
@tabchas what is your OS?

tabchas, 2016-06-04:
@ivankravets I am on OSX 10.11.5

ivankravets, 2016-06-06:
@tabchas thanks a lot for the report. Yes, the latest PlatformIO 2.9.4 should work very well on the Mac and Windows. 

We have the one issue with Linux - `udev` rules. User have to specify valid PID/VID `udev` rules for the SAM-BA based board. The one rule should be specified for the default Native USB port and other for the PROGRAM/BOOTLOADER serial port.

valeros, 2016-06-21:
Added support for several SAMD21 based boards:
```
mkr1000USB
adafruit_feather_m0_usb
sparkfun_samd21_dev_usb
sparkfun_samd21_mini_usb
```
Please install PlatformIO [dev version](http://docs.platformio.org/en/latest/installation.html#development-version) and update `framework-arduinosam` package via `platformio update`
Special thanks to @malachib for providing a test board!

ivankravets, 2016-06-21:
@malachib does it work for you?

malachib, 2016-06-22:
Using pio version 2.10.4.dev2 and the newest udev rules as of today.

Debian results are very positive.  Initial attempt at `pio run -t upload` yields a fully successful upload!  Subsequent attempts fail with:

`scons: *** [upload] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'
`

which can be overcome either by:

1.  running `sudo pio run -t upload` 
2.  waiting a certain amount of time (seems to be over a minute)

FYI I also tried this on an Ubuntu 16.04 box using the versions described on the first line.

The results are similar to my Jun 3rd post *except* that `sudo pio run -t upload` does seem to consistently upload to the device after a few tries.

ivankravets, 2016-06-22:
@malachib do you use this `board = sparkfun_samd21_dev_usb`? Check that 2 USB devices (in boot mode and without) are listed in udev rules

malachib, 2016-06-22:
I am indeed using `board = sparkfun_samd21_dev_usb`

I'll get back to you on the udev rules -

ivankravets, 2016-06-23:
@malachib do you have any news?

valeros, 2016-06-23:
Hi @malachib! Does upload process work well in Arduino IDE?
Probably this issue isn't linked with your udev rules. I've found some information about this problem on the [SparkFun website](https://learn.sparkfun.com/tutorials/samd21-minidev-breakout-hookup-guide/setting-up-arduino) (at the bottom of the page) and looks like it's a known issue with their board.

ivankravets, 2016-06-28:
@malachib I propose to open a new issue with detailed description about your problem. I'm going to make release with the other boards that described here.

closed 2016-06-28 by ivankravets

malachib, 2016-06-29:
Sorry been out of town!  I'll try out Arduino IDE and get back to you