platformio/platformio-core #521: Docs for Dash

sstaub, 2016-02-18:
I'm using dash ( https://kapeli.com/dash ) for the all the docs for language references etc.
It will a great help to use the docs for plattformio offline. Maybe you can support this app?

ivankravets, 2019-09-04:
@Kapeli what do you need from our side to have PlatformIO in Dash?

Kapeli, 2019-09-04:
> @Kapeli what do you need from our side to have PlatformIO in Dash?

I haven't seen any demand for PlatformIO within Dash yet, so it's unlikely I'll support it anytime soon.

If you want to, you can make and contribute a docset yourself: https://github.com/Kapeli/Dash-User-Contributions.

ivankravets, 2019-09-04:
@Kapeli thanks for the link! I hope someone from our community will do that. It's very difficult for us to maintain different documentation.

ivankravets, 2019-09-04:
@sstaub please move issue to https://github.com/Kapeli/Dash-User-Contributions/issues

closed 2019-09-04 by ivankravets