platformio/platformio-core #523: auto-generated CMakeLists.txt is not appropriate for VCS (clion)

boneskull, 2016-02-20:
I execute the following:

```shell
$ platformio init --ide clion --board d1_mini
```

`CMakeLists.txt` is generated.  It contains a lot of stuff, but also:

```
set(ENV{PATH} "/Users/boneskull/.nvm/v0.10.31/bin:/Users/boneskull/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Applications/Server.app/Contents/ServerRoot/usr/bin:/Applications/Server.app/Contents/ServerRoot/usr/sbin:/usr/local/git/bin:/usr/local/MacGPG2/bin:./node_modules/.bin:/usr/local/bin:/usr/local/sbin:/usr/local/heroku/bin:/usr/local/opt/coreutils/libexec/gnubin")
```

and references to every single library I have installed, regardless of whether or not my project uses them.

ivankravets, 2016-02-26:
Please re-test development version.

boneskull, 2016-03-01:
@ivankravets By this do you mean clone & install?

ivankravets, 2016-03-01:
@boneskull 
```
clone repo
platformio init --ide clion -b YOUR_BOARD
open CLion
```

I don't see here other normal solution. We will contact with CLion vendor and discuss "PlatformIO IDE for CLion".

The other hook: I can add new target to CLion named "Rebuild C/C++ Project Index" which will regenerate CMakeLists*.txt files.

WDYT?

ivankravets, 2016-03-07:
Please reopen this issue if you still have questions.

commit 52bbe9af (2016-02-26): Generate appropriate project for CLion IDE and CVS // Resolve #523
closed 2016-03-07 by ivankravets