platformio/platformio-core #524: IDE keep showing "Reload Now" popup at launch

rajendra2, 2016-02-20:
Hello,

I just started to use platformIO on mac OS X. I keep getting popup whenever I launch the IDE. It says Installing IDE... and after completion, there is option to Reload Now.

After choosing "Reload Now", it agains tries installing IDE and show option to Reload Now. This keeps happening and I can only get out of it if I choose , Reload later. I am guess this is not a normal behavior.  Any workaround ?

[image]
[image]

ivankravets, 2016-02-20:
I see it. Sorry. Give me a few minutes.

ivankravets, 2016-02-20:
Please sorry for that issue:

* Please click on "Reload Later"
* Then click on `gear` icon on PlatformIO IDE Toolbar which will open Atom's Settings
* Then go to `Settings > Packages` and upgrade PlatformIO IDE to 1.0.4.
* Restart IDE

rajendra2, 2016-02-20:
That was fast!

I have followed the steps and the popup is no longer coming after reloading. 

Thanks

closed 2016-02-20 by rajendra2