platformio/platformio-core #525: avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.

Vikaton, 2016-02-20:
I'm trying to build my arduino project and upload it, but when the build commands commence, I get an error

`avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.`

I tried running Atom itself as admin but that didn't seem to do the trick, any pointers?

ivankravets, 2016-02-20:
Did you close all Terminal windows with Serial Port Monitor?

Vikaton, 2016-02-20:
@ivankravets That seemed to do the trick, thanks! I had the arduino IDE Monitor open

closed 2016-02-20 by Vikaton

ela21ccc, 2017-04-02:
what do you mean by closing the serial monitor?

ivankravets, 2017-04-02:
@ela21ccc just reconnect your board

Mohit2183, 2018-06-09:
i am getting this error "avrdude: ser_open(): can't open device "\\.\COM4": Access is denied." what to do , i have tried with reconnecting it too

ivankravets, 2018-06-09:
It seems that this port is used by another other application. Try to disconnect it from other apps.

Mohit2183, 2018-06-09:
thanks ivankravets

jmcdaris, 2019-10-24:
I'm getting this same error.  I think I've done all the things mentioned above and still no luck.  What else can I try?

Rultem, 2019-11-07:
jmcdaris, Did you solve it?

tophat17, 2020-02-04:
I have no idea why but my issue when away after I closed Cura slicer. Make sure you don't have any programs on your computer open that eat all your ports.

chubylive, 2020-02-15:
Yeah Cura hugs your serial port like no other. you can disable the USB printer plugin

stadekawa, 2020-06-14:
Took too long to find this solution; closing Cura solved my issue. Thank you!

PhilRabbittBJ, 2020-09-06:
Yep Cura was the culprit for me too :-)
Thanks.

Lookingformybrother, 2021-04-21:
I tried to disconnect everything, close IDE and connect and open IDE again, but didn't work. for me it was Tools->Port -> and pick the port that is available there :)

locrider246, 2022-04-21:
> Reference

Thank you. I had the same error and I solved it thanks to your instruction.