platformio/platformio-core #526: Defining variables on the command line breaks for strings.

skorokithakis, 2016-02-21:
I tried to define a variable on the command line (a string), but while the `#define FOO "bar baz"` works fine in the code, I couldn't get it to work on the console (strings got split no matter what I tried).

I can post a minimal example soon, but it's simple enough to reproduce, just take a string define out of the file and into the command line. If the define contains spaces, building will break.

ivankravets, 2016-02-21:
How about ?
```ini
[env:myenv]
build_flags = '-DFOO="bar\ baz"'
```

skorokithakis, 2016-02-21:
I don't want to specify it there because my goal is to release an open source repo without my secret keywords (WiFi password) in it. So, I added a build.sh to my gitignore that specifies these options on the command line so secrets won't leave my computer.

Since this doesn't work, I'm forced to keep a public and a private version of a repo, which is cumbersome :-(

skorokithakis, 2016-02-21:
Oh, sorry, you meant escaping the space. I'll try that soon, ghank you!

ivankravets, 2016-02-21:
It should work. Please don't close this issue while we will not resolve it. Thanks.

skorokithakis, 2016-02-21:
No :(

~~~
export PLATFORMIO_SRC_BUILD_FLAGS='-DWIFI_PASS="my\ ssid" -DWIFI_SSID="MyWIFI"'

platformio run --target upload
~~~

~~~
#ifndef WIFI_PASS
#define WIFI_PASS "your wifi password"
#endif
~~~

~~~
main.ino:410:1: error: stray '\' in program
main.ino:410:1: error: stray '\' in program
main.ino:410:1: error: stray '\' in program
main.ino:410:1: error: stray '\' in program
<command-line>:0:11: error: 'MyWIFI' was not declared in this scope
main.ino:410:12: note: in expansion of macro 'WIFI_SSID'
~~~

Looks like it happens even for things with no spaces, sorry for the mistaken report, but it's worse than I thought.

skorokithakis, 2016-02-21:
If you can tell me where in the code this is handled, I can take a look at fixing it.

ivankravets, 2016-02-21:
Try 
```
export PLATFORMIO_SRC_BUILD_FLAGS="'-DWIFI_PASS=\"my\ ssid\"' -DWIFI_SSID=MyWIFI"
```

My output
```
(develop) ikravets@innomac /V/S/P/G/p/origin> platformio run -d examples/wiring-blink/ -e uno
[Sun Feb 21 17:10:29 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10607 "-DWIFI_PASS="my\ ssid"" -DWIFI_SSID=MyWIFI -DPLATFORMIO=020804 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant src/main.cpp
```

skorokithakis, 2016-02-21:
That still doesn't work, but this does:

~~~
export PLATFORMIO_SRC_BUILD_FLAGS="'-DWIFI_PASS=\\\"\"my\ ssid\"\\\"' '-DWIFI_SSID=\\\"\"MyWIFI\"\\\"'"
~~~

Wow. That is extremely convoluted. Is there any way PlatformIO command-line options for this could be simplified?

valeros, 2016-02-21:
Could you try please:
`export PLATFORMIO_SRC_BUILD_FLAGS='-DWIFI_PASS=\"MY WIFI KEY\"' '-DWIFI_SSID=\"TEST STR\"'`

skorokithakis, 2016-02-21:
That doesn't work, unfortunately. The compiler command line becomes `-DWIFI_PASS="MY`.

ivankravets, 2016-02-22:
@skorokithakis I'll look on it and fix from the PlatformIO's side

valeros, 2016-02-24:
@skorokithakis Could you please test [develop](http://docs.platformio.org/en/latest/installation.html#development-version) version of PlatformIO with these defines:
for bash:
`export PLATFORMIO_SRC_BUILD_FLAGS='-DWIFI_PASS="TEST PASS" -DWIFI_SSID="SSID NAME"'`
for Win cmd:
`SET PLATFORMIO_SRC_BUILD_FLAGS=-DWIFI_PASS="TEST PASS" -DWIFI_SSID="SSID NAME"`

skorokithakis, 2016-02-24:
Nope, this breaks integers:

~~~
build_flags = -DMQTT_MAX_PACKET_SIZE=1024
xtensa-lx106-elf-gcc -DMQTT_MAX_PACKET_SIZE=\"1024\"
~~~

closed 2016-02-25 by valeros

skorokithakis, 2016-02-25:
Except for the forgotten print statement, that worked, thanks!

valeros, 2016-02-25:
@skorokithakis Please test examples http://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_BUILD_FLAGS

skorokithakis, 2016-02-25:
I did and they work, thanks! Just the print statement left to take out.

asukiaaa, 2019-04-14:
I need to use `\\\` to define string variables.

```
PLATFORMIO_BUILD_FLAGS="-D DEVICE_ID=\\\"a_device_id\\\"" pio run
```
```
SUCCESS
```

but

```
PLATFORMIO_BUILD_FLAGS="-D DEVICE_ID=\"a_device_id\"" pio run
```
```
src/main.cpp: In function 'void loop()':
<command-line>:0:11: error: 'a_device_id' was not declared in this scope
src/main.cpp:16:18: note: in expansion of macro 'DEVICE_ID'
Serial.println(DEVICE_ID);
^
Compiling .pioenvs/leonardo/FrameworkArduino/HardwareSerial0.cpp.o
*** [.pioenvs/leonardo/src/main.cpp.o] Error 1
```

My environment
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

$ python --version
Python 2.7.15rc1

$ pio --version
PlatformIO, version 3.6.6
```

Tested project.
https://github.com/asukiaaa/pio_macro_practice

Is there any configuration to reduce number of `\`?

commit 1f1dc9c5 (2016-02-24): Better handling of build flags passed via environment vars // Resolve #526
commit 1c481e08 (2016-02-25): Fix handling of build flags passed via environment vars // Resolve #526